NOCIONS/letswave6 #2: Lets6 stuck when importing Lets5 files

EVale, 2015-03-30:
30/03/2015
I first updated Lets6 and then attempted to import averaged merged single subjects files (4 files 18 subjects each) into lets6 but it stucks at the first file.
Let me know if I need to send any file.

Best,

elia

amouraux, 2015-03-30:
Can you copy-paste the error message?

Best wishes,
André

EVale, 2015-03-30:
Dear André,

I tried now on a different computer and it worked. However, when I open the
files I see"empty" files

[image: Inline images 1]

In addition, when I click on the topright x to close them they do not close.

PLease note that yesterday I attempted to import having two instances of
matlab opened (ith Lets5 and 1 with Lets6).
This may have had a significant effect...

closed 2015-04-10 by amouraux