NOCIONS/letswave6 #11: Update troubles

EVale, 2015-04-29:
Dears,

I tried to update from private network (home) and got this message:

??? Error while evaluating uicontrol Callback

??? Error using ==> urlwrite at 140
Error downloading URL. Your network connection may be down or your proxy settings improperly configured.

Error in ==> CGLW_update>check_update_btn_Callback at 228
urlwrite(fullURL,filename);

Error in ==> gui_mainfcn at 96
        feval(varargin{:});

Error in ==> CGLW_update at 24
    gui_mainfcn(gui_State, varargin{:});

Error in ==> @(hObject,eventdata)CGLW_update('check_update_btn_Callback',hObject,eventdata,guidata(hObject))

As far as I see my network works properly and I'm connected with cable (ethernet - adsl)

amouraux, 2015-04-29:
Dear Elia,

If you have not updated in a while, you should download the archive from the website and update manually, as we have moved the files on Github, a public repository. This makes it now much easier for people to update and contribute to the project.

Best wishes,
André

EVale, 2015-04-29:
sure!

I will!

thanks,

e

closed 2015-05-07 by amouraux