NOCIONS/letswave6 #12: Update issue

hughherald, 2015-05-06:
[image]

When we try to update letswave, we receive this error message and you could notice that the new files are not properly located. (path problem)

Gan & Cedric.

amouraux, 2015-05-07:
Finally understood why I could not replicate this problem on my computer. It is due to the fact that the folder where LW6 is installed has the same name as the temporary folder where the ZIP is unzipped. Trying to find a workaround.

amouraux, 2015-05-07:
Fixed! Can you test it?

amouraux, 2015-05-07:
You will have to manually update first of course...

hughherald, 2015-05-07:
Yes, I tested it. everything fine!

Thanks.

closed 2015-05-07 by hughherald