NOCIONS/letswave6 #13: Error running ICA - datasets do not have matching unmixing matrices

GiuliaL, 2015-05-06:
Hi Andre,
When applying the ICA filter right after computing the ICA matrix (merged datasets option), I get the following error:

ERROR : the datasets do not have matching unmixing matrices

It is strange, because this happens only with the data from my last subject (I think the only one pre-processed 100% with Letswave6). With all my other data, the ICA filter works perfectly, and I don't think I am doing anything different. Do you have any idea what the problem could be?

Thanks in advance!

Giulia

amouraux, 2015-05-07:
Can you check whether the problem persists? I cannot reproduce the problem on my setup, even when using your files. Note that there was a bug in the update function. I just updated manually the LW6 files on your computer.

GiuliaL, 2015-05-07:
I still got the problem when applying the filter, so I re-ran the whole ICA from scratch and it finally worked.
Thanks!

closed 2015-05-07 by GiuliaL