NOCIONS/letswave6 #14: assign file and apply ICA

cedlenoir, 2015-05-11:
When I build the matrix on files (epochs 3s) and perform the ICA to remove eyes movements : no issue, but when I try to assign one of the ica files to others files (epochs 0.2s) I get the following error message, and the interface window of ICA appear very briefly.

Error using  == 
Matrix dimensions must agree.

Error in GLW_ICA_apply_filter>GLW_ICA_apply_filter_OpeningFcn (line 128)
            if tp_um==ICA_um;

Error in gui_mainfcn (line 221)
    feval(gui_State.gui_OpeningFcn, gui_hFigure, [], guidata(gui_hFigure), varargin{:});

Error in GLW_ICA_apply_filter (line 22)
    [varargout{1:nargout}] = gui_mainfcn(gui_State, varargin{:});

Error in LW_ICA_apply_filter (line 141)
        [a out_configuration]=GLW_ICA_apply_filter('dummy',configuration,datasets);

Error in letswave_gui>menu_callback (line 329)
                configuration=eval(eval_st);
 
Error while evaluating uimenu Callback

amouraux, 2015-05-19:
Can you check whether the datasets have the exact same number of channels?

cedlenoir, 2015-05-19:
They have the same number of channels, the difference is the length of the epochs.

cedlenoir, 2015-05-19:
After the correction you made to assign the matrix, the same problem remains when I try : ICA apply spatial filter on the files on which the matrix was assigned.
Error using  == 
Matrix dimensions must agree.

Error in GLW_ICA_apply_filter>GLW_ICA_apply_filter_OpeningFcn (line 128)
            if tp_um==ICA_um;

Error in gui_mainfcn (line 221)
    feval(gui_State.gui_OpeningFcn, gui_hFigure, [], guidata(gui_hFigure), varargin{:});

Error in GLW_ICA_apply_filter (line 22)
    [varargout{1:nargout}] = gui_mainfcn(gui_State, varargin{:});

Error in LW_ICA_apply_filter (line 141)
        [a out_configuration]=GLW_ICA_apply_filter('dummy',configuration,datasets);

Error in letswave_gui>menu_callback (line 329)
                configuration=eval(eval_st);
 
Error while evaluating uimenu Callback

amouraux, 2015-06-24:
Can you update and try again?

closed 2015-12-04 by amouraux