NOCIONS/letswave6 #26: Gui options error

EVale, 2015-06-08:
Master Version downloaded on 8/06/15

??? Error: File: letswave_GUI_options.m Line: 67 Column: 18
A function name ("config") cannot be indexed with ".".

Error in ==> letswave_gui>proc_gui_options_Callback at 1154
letswave_GUI_options;

Error in ==> gui_mainfcn at 96
        feval(varargin{:});

Error in ==> letswave_gui at 34
    gui_mainfcn(gui_State, varargin{:});

Error in ==> @(hObject,eventdata)letswave_gui('proc_gui_options_Callback',hObject,eventdata,guidata(hObject))

 
??? Error while evaluating uimenu Callback

***Let me know if you need more info***

amouraux, 2015-06-09:
Dear Elia,

It seems that it cannot find the letswave.mat which should be present in your letswave6 folder.
Have you added the path to the letswave6 folder?
Is that letswave.mat folder present?

I just changed something which, hopefully, will fix the error.

Best wishes,
André

EVale, 2015-06-09:
Dear André,

I actually received several other errors with other functions. I tried the
original version and it worked better.
I usually add path pressing adding folders (without subfolders). Not sure
if this may affect.

Let me know if you want me to test anything...

best

e

amouraux, 2015-06-09:
Dear Elia,

Can you update again and send me an example error message?
Yes, the only folder you need to add is the letswave6 folder (without the subfolder).
Also, maybe check that you do not have the folder/subfolders of letswave5 in your path, as this could mess things up…

Best wishes,
André

EVale, 2015-06-09:
Dear André,

I redownloaded, deleted old versions paths and lets5 paths.
Now it seems stable.
If I get other warnings I'll post you.

Thanks!!

closed 2015-06-13 by EVale