NOCIONS/letswave6 #27: anova

EVale, 2015-06-12:
Dear André,

I've got this message:

Starting matlabpool using the 'local' profile ... connected to 2 labs.
Total number of tests to perform : 30060
C:1 Z:1 Y:1
Error using parallel_function (line 589)

CAT arguments dimensions are not consistent.

Error stack:
ILW_custANOVA_parfor>(parfor body) at 116

Error in ILW_custANOVA_parfor (line 112)
            parfor dy=1:header.datasize(5);

Error in RLW_ANOVA (line 131)
    [out_dataset_pvalue,out_dataset_Fvalue]=ILW_custANOVA_parfor(datasets,within_levels,within_labels,between_levels,between_labels);

Error in LW_ANOVA (line 82)
        [out_datasets(1),out_datasets(2),message_string,cluster_distribution]=RLW_ANOVA(datasets,configuration.parameters.factor_data,configuration.parameters.table_data,'enable_parallel',configurati
        Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating uimenu Callback

May be due to the data I have input?
Let me know if you want me to send them.
I have input four merged single subj avg files (26 epochs each).

EVale, 2015-06-12:
this happens if I do not ask for parallel computing:

Error using vertcat
CAT arguments dimensions are not consistent.

Error in ILW_custANOVA (line 118)
                    tpdata=[tpdata;squeeze(datasets(k).data(:,channelpos,1,dz,dy,:))];

Error in RLW_ANOVA (line 133)
    [out_dataset_pvalue,out_dataset_Fvalue]=ILW_custANOVA(datasets,within_levels,within_labels,between_levels,between_labels);

Error in LW_ANOVA (line 82)
        [out_datasets(1),out_datasets(2),message_string,cluster_distribution]=RLW_ANOVA(datasets,configuration.parameters.factor_data,configuration.parameters.table_data,'enable_parallel',configurati
        Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating uimenu Callback

I will check if there is anything wrong with some datafile...

amouraux, 2015-06-12:
Yes. Or you can send me the data sets and anova model!

EVale, 2015-06-12:
I FOUND AN ERROR. Sorry I thought it had to do something with parallel computing. This last version seems pretty stable!!! Well done!!!!!! (I was quite happy with Lets5 but I will accept its departure :-) ) :+1:

closed 2015-06-12 by EVale