NOCIONS/letswave6 #30: MEG sensors location and headmodel estimation

mqwhe, 2015-06-22:
Hi there,
I'm trying to import MEG data into Letswave and so far the sensor location and headmodel computation needs more work. I converted the KIT-MEG 160 sensor coordinate into the XYZ ‘chanlocs’ file, but the output seems weird (below image). I'm wondering whether it's because the native MEG sensor is elliptic shaped and Letswave can only deal with "uniform radius” layout?  The distance information is important in case of MEG while not so serious in EEG? However, other EEG software, e.g., SPM and EEGlab, seem to have no difficulty in accommodating MEG sensor co-registrations. Any comments/suggestions is very much appreciated and please let me know if you need me to send through an expampler data. 
[image]

amouraux, 2015-06-25:
Strange that eeglab imports the locations correctly, but not letswave, as both rely on the same function. Can you send me the file to have a look?

Andre

mqwhe, 2015-06-26:
Hi Andre, 
Thanks for your reply. How to send you the files? 
Cheers,
Wei

amouraux, 2015-06-26:
Hello,

I guess you can send it to me by email, at andre.mouraux@uclouvain.be <mailto:andre.mouraux@uclouvain.be>

André

closed 2018-03-12 by amouraux