NOCIONS/letswave6 #35: Analyse data from accelerometer

gualoe, 2015-08-12:
Hello,
we would like to use letswave to preprocess accelerometer data, using just one channel. I've tried to upload a simple ascii file (no header information) specifying sampling rate etc., but it doesn't work,

Best wishes,
Guido

amouraux, 2015-08-21:
Dear Guido,

Normally, it should work. 
Can you give me a sample ASCII file so that I can test it?

Of note, I remember that I fixed a bug regarding the ASCII import a few weeks ago. Can you make sure that you have the latest updated version?

Best wishes,
André

gualoe, 2015-08-26:
Dear Andre,

apologies for the delay, I have been annual leave, please find attached the relevant file in .csv and ascii format, the sampling rate is 0.5 Hz. The file does not actually contain raw acceleration data, but reflects cross recurrence of acceleration. Best, Guido

 
………………………………………...
Dr Guido Orgs
Lecturer in Psychology
Department of Life Sciences
College of Health and Life Sciences
Brunel University London

Email: guido.orgs@brunel.ac.uk <mailto:guido.orgs@brunel.ac.uk>
Phone: +44(0)18952 67707
 www.brunel.ac.uk

> 
> Dear Guido,
> 
> Normally, it should work. 
> Can you give me a sample ASCII file so that I can test it?
> 
> Of note, I remember that I fixed a bug regarding the ASCII import a few weeks ago. Can you make sure that you have the latest updated version?
> 
> Best wishes,
> André
> 
> > 
> > Hello,
> > we would like to use letswave to preprocess accelerometer data, using just one channel. I've tried to upload a simple ascii file (no header information) specifying sampling rate etc., but it doesn't work,
> > 
> > Best wishes,
> > Guido

62.90046296
60.70245726
60.40277778
60.42592593
60.36841168
60.2647792
60.88657407
60.51834046
60.41452991
59.45868946
36.09455128
66.59757835
61.86538462
61.54291311
61.64618946
61.24341168
61.58903134
61.40473647
60.99038462
61.40473647
61.23201567
61.30110399
61.38158832
61.50854701
61.71599003
61.53116097
61.93429487
61.85363248
61.55430912
61.40491453
61.46225071
61.19729345
61.23219373
61.23201567
61.3468661
61.61182336
[... 957 lines cut ...]
60.05769231
60.72542735
61.04807692
60.58707265
60.44889601
60.84027778
59.93108974
58.83725071
60.74893162
60.88621795
60.86360399
58.79095442
62.15295584
61.94604701
53.49358974
56.52225783
51.26086182
49.59152422
60.95530627
61.6235755
60.57549858
58.42289886
59.28632479
61.49697293
61.23254986

amouraux, 2015-10-20:
Dear Guido,

I made some additional fixes to the import ASCII functions. 
It should now work fine?

Best wishes,
André

gualoe, 2015-10-20:
> Email: guido.orgs@brunel.ac.uk<mailto:guido.orgs@brunel.ac.uk> <mailto:guido.orgs@brunel.ac.uk>

Dear Andre,

thanks for looking into this. I’m currently switching institutions and as yet without matlab access but will check soon,

Best,
Guido

-----------------------------------------------
Dr Guido Orgs
Lecturer, Department of Psychology
Goldsmiths, University of London

Whitehead Building, Room W200/6
New Cross, London, SE14 6NW

E-mail: g.orgs@gold.ac.uk<mailto:g.orgs@gold.ac.uk>
Phone: +44 (0)20 7896 2719

closed 2018-03-12 by amouraux