NOCIONS/letswave6 #47: LW5 plugins for LW6

JoanLS, 2015-11-03:
The modified plugins from LW5 do not appear with the latest update of LW6?

amouraux, 2015-11-03:
No, because Bruno did not want them to be included in LW6.
You can download them from the LW6 github wiki for now.
Then, we will see whether we can host them somewhere with some protection…

Best wishes,
André

JoanLS, 2015-11-04:
Thanks André! 

Apologies for spamming but I have found a couple more plugins from LW5 that would need to be converted. 

https://dl.dropboxusercontent.com/u/20128470/To%20Add.zip

I am not 100% sure of the authors, but I think it's as follows

- GLW_interpolate_channels_of1epoch by Dan Nemdorov
- GLW_manage_electrodes by Gilles Vertongen
- GLW_normalize_topographies by Corentin Jacques
- GLW_segment_recording_blocks by Dan Nemdorov
- GLW_sumepochs by Dan Nemdorov
- GLW_XYZnorm by Corentin Jacques
- LW_PreprocessDataNancy by Goedele Van Belle

Also, there are a couple of plugins currently listed that were not created by Corentin but by Goedele
- GLW_conditionAverage
- GLW_peakPreview

As always, thank you very much in advance for your help!

Best,
Joan

amouraux, 2015-12-02:
Ok. Will try to add them ASAP!
André

amouraux, 2015-12-04:
For the plugin GLW_interpolate_channels_of1epoch, I am missing the file LW_interpolate_channels_of1epoch.m

JoanLS, 2015-12-09:
Dear André,

It seems that this plugin is not fully developed (Dan created it I think)
and nobody in the lab has the .m file. We can leave out this plugin for LW6.

Thank you very much,
Joan

amouraux, 2015-12-12:
Dear Joan, 

I added some additional plugins. 

There are still a couple of plugins that I cannot convert because of missing functions : 

GLW_normalize_topographies
- Missing LW_norm_topos

GLW_manage_electrodes
- Missing LW_switch_electrodes

GLW_sumepochs
- Missing LW_sumepochs

LW_preprocess_data_Nancy
- Explanations required as to what this function does

JoanLS, 2015-12-15:
Dear André,

Please find the missing functions in attachment.

As for* LW_preprocess_data_Nancy*, this is a plugin for preliminary
analysis of intracerebral data. It mainly integrates pre-existing functions
from Letswave in a GUI.
-  imports data in both .TRC format (Micromed recording format) and
Brainvision .edf format

- search for events (digital triggers), in absence of events, select an EEG
channel to use as analog trigger via "create events from level trigger" (in
our case, we have the channel *MKR1+* recording the photodiode)

- visualise recording segments with triggers and select relevant blocs
(with possibility to manually move start & end of blocs)

- import stimulation information from Sinstim output (select Sinstim .mat
outputs, then select .txt file with trial conditions)

- crop and run FFT on selected EEG segments

I have included a .zip file with an example TRC file and the Sinstim
output, so you can see what it looks like :*
https://dl.dropboxusercontent.com/u/20128470/LW_PreprocessNancy_Example.zip
<https://dl.dropboxusercontent.com/u/20128470/LW_PreprocessNancy_Example.zip>*.
In this experiment, images are shown at 6 Hz, with alternating scrambled &
face images (so frequency of interest = 3 Hz).

Have a nice day,
Joan

function [outheader,outdata] = LW_sumepochs(header,data,varargin)
% LW_sumepochs
% Sum epochs
%
% Inputs
% - header : LW5 header
% - data : LW5 data
% - varargin : weightvector
%   Unweighted averaging : 
%   [outheader,outdata]=LW_sumepochs(header,data);
%   Weighted averaging :
%   [outheader,outdata]=LW_sumepochs(header,data,weightvector);
%
% Outputs
% - outheader : LW5 header
% - outdata : LW5 data
%
% Dependencies : none
%
% Author : 
% Talia Retter, modifcation of line 59 of "LW_sumepochs" function by:
% Andr? Mouraux
% Institute of Neurosciences (IONS)
% Universit? catholique de louvain (UCL)
% Belgium
% 
% Contact : andre.mouraux@uclouvain.be
% This function is part of Letswave 5
% See http://nocions.webnode.com/letswave for additional information
%

%determine the weight vector, use ones if not weighted averaging
tp=size(varargin,2);
if (tp>0);
    weight=varargin{1};
else
    weight=ones(size(data,1),1);
end;
[... 133 lines cut ...]

normMethod = lower(normMethod);

%loop through all the data. 
for epochpos=1:size(data,1)    
        for indexpos=1:size(data,3)
            for dz=1:size(data,4)
                for dy=1:size(data,5);
                    switch normMethod
                        case 'mccarthy & wood'                            
                           scaling =  sqrt(sum(squeeze(data(epochpos,selectedchannels,indexpos,dz,dy,:)).^2,1));
%                            scaling =  sum(abs(squeeze(data(epochpos,selectedchannels,indexpos,dz,dy,:))),1);
                           disp(num2str(scaling));
                           outdata(epochpos,:,indexpos,dz,dy,:) = bsxfun(@rdivide,squeeze(data(epochpos,selectedchannels,indexpos,dz,dy,:)),scaling);
                            
                        case 'z-score';
                            outdata(epochpos,:,indexpos,dz,dy,:) = zscore(squeeze(data(epochpos,selectedchannels,indexpos,dz,dy,:)));
                            
                        otherwise
                            error('unknown normalize method!');                            
                    end
                end
            end
        end
end

amouraux, 2015-12-15:
Dear Joan,

Typically, what this function does could now be implemented as a script?

Best wishes,
André

JoanLS, 2016-01-18:
Hi André,

Apologies for the late reply. The LW_preprocess_data_Nancy plugin is useful
for also visualising the triggers so we can easily evaluate the stimulation
sequences. It is also useful to be able to import the Sinstim output to
extract the exact stimulation frequencies and rename the triggers according
to the stimulation conditions.

Perhaps if there were LW6 functions that could do these things we could
bypass the plugin and use scripting instead?

Have a nice day,
Joan

closed 2018-03-12 by amouraux