NOCIONS/letswave6 #51: RLW_merge_channels.m, line 70

BaptisteChemin, 2015-11-18:
Hello,

I had an error with the function RLW_merge_channels.m, line 70:

Error using horzcat
Number of fields in structure arrays being concatenated do not match. Concatenation of structure arrays requires that these arrays have the same set of fields.​

The problem is due to the fact that the different datasets from which I want to merge the channels do not always have electrodes coordinates. 

For now, I fixed it like this:

        %chanlocs
        if length(fieldnames(out_header.chanlocs)) > length(fieldnames(header.chanlocs))
            out_header.chanlocs = rmfield(out_header.chanlocs, {'theta', 'radius', 'sph_theta', 'sph_phi', 'sph_theta_besa', 'sph_phi_besa', 'X', 'Y', 'Z'});
        elseif length(fieldnames(out_header.chanlocs)) < length(fieldnames(header.chanlocs))
            header.chanlocs = rmfield(header.chanlocs, {'theta', 'radius', 'sph_theta', 'sph_phi', 'sph_theta_besa', 'sph_phi_besa', 'X', 'Y', 'Z'});
        end
        out_header.chanlocs=[out_header.chanlocs header.chanlocs];

Unfortunately, this removes the info about electrodes coordinates, and suppose that the supplementary fields in header.chanlocs are always the same... I'm sure there should be a way to deal with this error more efficiently (by adding empty fields instead of removing the "supplementary fields"?), but so far I did not succeed doing it. Any idea?

Best,

Baptiste

amouraux, 2015-11-26:
Fixed! Thanks!
André

closed 2015-11-26 by amouraux

BaptisteChemin, 2015-11-27:
Bonjour,

J'ai de nouvelles erreurs avec la fonction "merge channels of different datasets":

Subscripted assignment between dissimilar structures.

Error in RLW_merge_channels (line 116)
            out_header.chanlocs(i)=tp3;

Error in LW_merge_channels (line 77)
        [out_datasets.header,out_datasets.data,message_string]=RLW_merge_channels(datasets,configuration.parameters.merge_idx);

Error in letswave_gui>menu_callback (line 387)

Voila des données utilisées et qui génèrent l'erreur:

https://www.dropbox.com/sh/9dbu8ev221geakr/AAC_2Q0uwoXmyxhxAjTmUa4qa?dl=0

(64 cannaux EEG (avec scalp topo) dans un dataset, 2 canaux auxiliaires (sans topo) dans l'autre).

Merci!

Baptiste