NOCIONS/letswave6 #58: printing mistake in permutation test for ttest

hughherald, 2016-01-04:
Dear Andre,
I found there is a mistake in RLW_ttest.m and RLW_ttest_constant.m,
which is just a printing mistake, but will make the results incorrectly.
I fixed it and committed it on Jan 4, 2016, see: https://github.com/NOCIONS/letswave6/commit/90508da96d8073aab7f000f554b413f9f856d090

For RLW_ANOVA.m, it seems also have some problem.
Should the code in line 209,
```matlab
tpsize(i)=sum(sum(tp_Fvalues(find(tp2==i))));
```
be 
```matlab
tpsize(i)=sum(sum(abs(tp_Fvalues(find(tp2==i)))));
```

Please check whether the change is correct or not.

amouraux, 2016-01-05:
Dear Gan,

Thank you for correcting this *BIG* mistake in RLW_ttest! 
This was definitely going to mess up the permutation testing!

Regarding the RLW_ANOVA. aren’t F values always positive?

Best wishes,
André

hughherald, 2016-01-05:
Yes, you are right. So no mistake in RLW_ANOVA.

closed 2016-01-05 by hughherald