NOCIONS/letswave6 #59: Importing EGI NetStation files

mmoulson, 2016-01-26:
Hello,
I'd like to use Letswave to process EEG data collected with the EGI system. Is there a way to import raw NetStation files, or is exporting to EDF+ the way to do it?
Thanks!
Meg
--
Margaret Moulson, PhD
Associate Professor, Ryerson University
Visiting Professor, UC Louvain

amouraux, 2016-01-27:
Hello, 
This is not possible for now. However, because the matlab import function of Fieldtrip supports EGI files supports EGI net station files, it should be very easy for us to implement. Can you send me a small sample dataset (e.g. via dropbox) so that I can test the function?

mmoulson, 2016-01-27:
Hello,
I have shared a file with you via dropbox. Let me know if you have issues accessing it. The file was a straightforward ERP recording session, 500Hz sampling rate. Let me know if you need other details about it (e.g., number of events, etc.). 
Thank you!

amouraux, 2016-01-30:
Thank you. 

I have just added a function to import *.egis, *.sbin and *.mff files in LW6.
However, I am unable to import your dataset.

Best wishes,
André

mmoulson, 2016-02-03:
Hi Andre,
Thanks for working on this! The dataset I sent you was collected in a later version of NetStation whose native format is not .egis. Earlier versions of NetStation saved data in .egis format. I've exported one of our files to .egis format and shared that with you via dropbox. I tried importing it using the new import function you created for NetStation files, but received this error:
Undefined function 'read_egis_header' for input arguments of type 'char'.

Error in ft_read_header (line 644)
    [fhdr,chdr,ename,cnames,fcom,ftext] = read_egis_header(filename);

Error in ft_read_data (line 114)
  hdr = ft_read_header(filename, 'headerformat', headerformat);

Error in RLW_import_EGI (line 42)
dat=ft_read_data(filename);

Error in LW_import_EGI (line 87)
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_EGI(filename,'concatenate',configuration.parameters.concatenate);
            
Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating uimenu Callback

Thanks,
Meg

amouraux, 2016-02-04:
Dear Meg,

I added the missing functions. You should now be able to import EGIS datasets. 
You might also be able to import the later version Netstation files. Can you try?

André

mmoulson, 2016-02-04:
Hi André,

I was able to import the EGIS dataset, which is great news. Thank you! I tried a later-version NetStation file though, and received the following error message. 

Warning: could not determine filetype of /Users/meg/Desktop/fpvs mat
files/ORE137test 
> In ft_filetype at 1002
  In ft_read_data at 65
  In RLW_import_EGI at 42
  In LW_import_EGI at 87
  In letswave_gui>menu_callback at 387 
Error using ft_read_header (line 1635)
unsupported header format (unknown)

Error in ft_read_data (line 114)
  hdr = ft_read_header(filename, 'headerformat', headerformat);

Error in RLW_import_EGI (line 42)
dat=ft_read_data(filename);

Error in LW_import_EGI (line 87)
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_EGI(filename,'concatenate',configuration.parameters.concatenate);
            
Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating uimenu Callback

Thanks very much for working on this. 
Meg

mmoulson, 2016-05-25:
Dear André,

A while ago, you created an import function for EGI NetStation .egis files (thank you!). I was wondering whether it would be possible for you to create a file with the channel locations for the EGI HydroCel 128-channel sensor net, so that we can assign electrode coordinates and create head plots for our EGI files. I found this on egi's ftp site:
ftp://ftp.egi.com/pub/support/3rdPartySoftwareSupport/BESA/BESA_sensor_position_files_HydroCelGSN/GSN-HydroCel-128.sfp
but I don't know if that's exactly what you need. Let me know if you need a particular file format and I can contact EGI support.

Thank you,
Meg

amouraux, 2016-05-26:
Actually, 

You should be able to use directly the SFP electrode coordinates… 

Best wishes,
André

mmoulson, 2016-06-15:
Hi André,

Yes, I was able to use the .sfp electrode coordinate file to assign coordinates in Letswave. Although this works on my computer (a Mac) and I have no problem then viewing the files using the waveform multiviewer, my student is having issues when she tries to do this on her PC. I have attached a screenshot of the error message we are getting.
[image]

If you have any insight about where we're going wrong, I would appreciate it.

Thanks,
Meg

amouraux, 2016-06-20:
Dear Meg,

This is not due to the operating system, but to the fact that it calls a function that requires the Statistics (or other) toolbox. 
I will see whether we could use another function to generate the head plots.

Best wishes,
André

closed 2018-03-12 by amouraux