NOCIONS/letswave6 #60: Issue with "segment epochs in successive chunks"

assafbeck, 2016-01-26:
Hello,
I don't get the good number of bins after chunking. 

Cheers,
Assaf

amouraux, 2016-01-27:
Hi,

I checked the function, but do not see any obvious error. 

Can you provide me with the parameters that you used, as well as the exact sampling rate of your dataset (or send me an example dataset)?

André

assafbeck, 2016-01-27:
Hello,

samplig rate: 127,9881
Chunk length (duration): 0.48805
Start chunking at (latency): 0.255975
Interval between chunks: 0.5
In fact I should have 41 bin whith my bin of interest on the 21th bin (it is the case on letswave 5) but I get 40 bins and my bin of interest is on the 22th

Thanks,
Assaf

amouraux, 2016-02-03:
Dear Assaf,

This must be a rounding issue. 

I am surprised that you expect (and get) 40-41 bins.

With a chunk length of 0.48805 and a sampling rate of 127.9881 Hz, I would expect the number of bins to be 0.48805*127.9881 = 62.4645922 bins?

André

amouraux, 2016-02-03:
Maybe that you can send me a small sample dataset so that I can investigate this further?

Best wishes,
André

assafbeck, 2016-02-03:
Hello,

I think the expect number of bins will be:
 chunk duration / frequentiel resolution = chunk duration/(frequency range/total numbers of bins) = numbers of bins
0.48805/(127.9881/10752) = 41.01 bins

Is that possible that the missing bin will be related to an automatic correction for missing first photodiode event ?

Example of dataset: https://www.dropbox.com/sh/0qq8jo8nd4im9gk/AAB_Hb0CvbkocNyDBFp3R3esa?dl=0.

Thank you,

Assaf

amouraux, 2016-02-04:
Problem should be fixed now. Simply update to use the revised functions. 

To convert the duration into a number of bins, I used the 'fix’ function instead of the ‘round’ function. 
However, the fix function takes the lowest integer value.

In your case, the chunk duration corresponded to : 
0.48805/0.11905=40.9962
fix(40.9962)=40 has now been replaced by round(40.9962)=41

Best wishes,
André

assafbeck, 2016-02-04:
Thank you very much !

Have a nice day,
Assaf

closed 2016-02-04 by amouraux