NOCIONS/letswave6 #61: Issue with "frequency spectrum SNR"

assafbeck, 2016-01-26:
Hello,
The number that I put to define the range of surrounding bins does not change the output.

Cheers,
Assaf

amouraux, 2016-01-27:
I just tested the function, and it does appear to change the output.
Are you certain that you have the latest version?
If so, can you provide me with a screenshot of the parameters entered in the GUI?
But maybe that you are referring to some plugin?

André

assafbeck, 2016-01-27:
Hello,
I don't know if I have the latest version, how can I tcheek for that?
I am referring to the function from "preprocess - baseline operation - 
frequency spectrum SNR"
I join a screenshot, the different parameters does not change the output

Thanks,
Assaf

amouraux, 2016-01-27:
You can check using the File > Check for updates function.
When did you last update?

assafbeck, 2016-01-29:
Hello,
I just did the update, but now my waveform multiviewer doesn't work 
anymore, I get an error message, I join a screenshot.

Thanks,
Assaf

amouraux, 2016-01-30:
It seems that screenshots are not included in the Github system. 
Can you copy-paste the error message?

André

assafbeck, 2016-02-01:
Hello,
here a copy of the error message.

Thank you,
Assaf

Error using load
Unable to read file init_parameter.mat: No such file or directory.

Error in CGLW_multi_viewer/init_parameter (line 19)
         S=load('init_parameter.mat');

Error in CGLW_multi_viewer/CGLW_my_view_OpeningFcn (line 477)
         init_parameter();

Error in CGLW_multi_viewer (line 15)
CGLW_my_view_OpeningFcn;

Error in letswave_gui>proc_multi_viewer_Callback (line 1019)
CGLW_multi_viewer(cb,inputfiles,send_update_status(handles));

Error in gui_mainfcn (line 96)
         feval(varargin{:});

Error in letswave_gui (line 34)
     gui_mainfcn(gui_State, varargin{:});

Error in 
@(hObject,eventdata)letswave_gui('proc_multi_viewer_Callback',hObject,eventdata,guidata(hObject))

Error while evaluating uimenu Callback

amouraux, 2016-02-01:
Hello,

Strange. It seems that the function cannot find the file “init_parameter.mat”. However, that file should be present in the ‘core functions’ folder of LW6. Either something went wrong when updating, or something is incorrect in your Matlab path settings. 

Could you delete the entire LW6 folder and reinstall from scratch? In your Matlab path settings, you should only add the parent folder, i.e. the folder where LW6 is installed (without the subfolders).

André

assafbeck, 2016-02-01:
Hello,

Actually I changed the path setting just with the LW6's file and not 
with the subfolders and its works.

I check the issue I had with the SNR and the number of bins to take into 
account, and it's fixed. However the issue with the chunk is still 
there, I miss one bin.

Assaf

closed 2018-03-12 by amouraux