NOCIONS/letswave6 #62: Problem plotting topographies

JoanLS, 2016-01-29:
I cannot plot EEG topographies even after assigning electrode coordinates & computing spline files. The resulting headplots only contain a couple of electrodes on the top of the head. There were no error messages during the electrode coordinate & spline file processing.
[image]

amouraux, 2016-01-30:
Dear Joan, 

The new topoplot functions no longer require to compute spline files. 
Can you check that you have the latest version?

If it still does not work, can you send me the file so that I can check whether the “assign electrode coordinates” worked correctly? Do the topoplot scalp maps look ok? If so, it is probably a small bug in the new head plot function which was written by Gan Huang...

André

JoanLS, 2016-02-01:
Hi André,

I have downloaded the latest version and the topoplot seems to be working
fine, but the 3D headplot is still bugging (see attached image).

I get the following error message :

Reference to non-existent field 'spl'.

Error in CGLW_multi_viewer/fig_headplot (line 1787)
                    set(
handles.dot_headplot(ax_idx),'XData',header.spl.newElect(:,1));

Error while evaluating uitoggletool ClickedCallback

An a small question : is it possible in the new multiviewer to block the
position of the cursor so it doesn't move with the mouse?

Thanks in advance!
Joan

Face Categorisation Lab <http://face-categorization-lab.webnode.com/>
IPSY - IoNS
Université Catholique de Louvain
Place Cardinal Mercier, 10
B-1348 Louvain-la-Neuve
Tel: 0032 10 47 92 60

JoanLS, 2016-02-01:
Hi André,

Please find attached a file with the working 2D but not 3D topographies.

I have also just noticed that the 2D topography will work on the
grand-averaged file, but not on the individual subject files (merged as
separate epochs). I have also included that file in the current e-mail.

Thank you for your help!

amouraux, 2016-02-02:
Can you send me the files in a separate email, or using a dropbox link? It seems that github scrubs all attachments… 

André

JoanLS, 2016-02-02:
Ah ok, here is a dropbox link :

https://dl.dropboxusercontent.com/u/20128470/Topo_Files_forAndre.zip

Face Categorisation Lab <http://face-categorization-lab.webnode.com/>
IPSY - IoNS
Université Catholique de Louvain
Place Cardinal Mercier, 10
B-1348 Louvain-la-Neuve
Tel: 0032 10 47 92 60

hughherald, 2016-02-03:
Dear JoanLS,

Thanks for report this issue. The new multi-viewer is developed quite new.
I've not test it so widely. Now  I'm trying to fix it.

For the 2D plot, for the default setup we are using "Standard-10-20-Cap81.locs", which would be assigned automatically.
In your case, there is lots of channels not in the "Standard-10-20-Cap81.locs", so you have to assigned it manually. The file "biosemi_locations_128.xyz" you using contains the channels information with the channel name A1,A2,A3,..., B1,B2,B3,..., C1,C2,C3.... I think it is not appropriate. So it leads to the problem of only a small part of channels are shown.

I think the correct file for this case is "BEM_standard_1005.elc" which is in the folder "...\letswave6\resources\electrodes\fieldtrip_standard_locations". But there are still 8 channel not be signed, which are 'PO11', 'POI1', 'POI2', 'PO12', 'EXG1', 'EXG2', 'EXG3', 'EXG4'. 
But I think the coordinate X,Y has been exchanged. So I edited it, and renamed it as "BEM_standard_1005_XYswap.elc". 

For 3D plot, there is a logical mistake produced by me. Now it is fixed.

Please update the letwave and check for any problem, especially for the coordinate.

JoanLS, 2016-02-03:
Hi Gan,

Thank you for your reply.

For the electrode names, we are using a 128-channel Biosemi system but we
had converted their standard naming (A1, A2...) into approximate 10-20
system names (Oz, O1, O2...) with a plugin called "Manage Electrodes".

This plugin allowed us to select multiple LW files, assign electrode
coordinates and spline files, and switch between the Biosemi - 10-20
electrode names. This was very practical for us as during preprocessing and
especially electrode interpolation, we could refer to the Biosemi
convention (more intuitive). After, for frequency-domain analysis, we could
switch to our adapted 10-20 names to compare to previous studies.

Would it perhaps be possible to adapt this plugin for LW6? All the files
are in the following dropbox folder :
https://dl.dropboxusercontent.com/u/20128470/LW_ManageElectrodes.zip

Thanks in advance!
Joan

Face Categorisation Lab <http://face-categorization-lab.webnode.com/>
IPSY - IoNS
Université Catholique de Louvain
Place Cardinal Mercier, 10
B-1348 Louvain-la-Neuve
Tel: 0032 10 47 92 60

hughherald, 2016-02-03:
Dear Joan,

I tried to change the channel names, then assign the electrodes by file "biosemi_locations_128.xyz".
The topography looks more reasonable than before. And also I can feel the necessary of the switch.

To the plugin,
I think there is not to much different between LW5 and LW6.
Assign the electrodes must be done after switch the channel name to standard naming (A1, A2...).
And making and assigning spl file is now not necessary. 
The multiviewer will generate it online every time when you load the data with a super fast speed.

Best,
Gan

amouraux, 2016-02-04:
Dear Joan, 

Is the problem fixed?

André

JoanLS, 2016-02-04:
Hi André & Gan,

I have tried Gan's solution and it now works well, thanks!

Is there any way of increasing the size of the 3D topoplot though? With 128
channels, it's not always easy to see where the peak channel is.

Face Categorisation Lab <http://face-categorization-lab.webnode.com/>
IPSY - IoNS
Université Catholique de Louvain
Place Cardinal Mercier, 10
B-1348 Louvain-la-Neuve
Tel: 0032 10 47 92 60

hughherald, 2016-02-06:
Dear Joan,

Actually, we have this design in the multiviewer.
As we designed, if you double click the topograph, a large figure will pop out, But...
 
But due to some other problem it does not work now.
I'll fix it soon.

Best,
Gan

EVale, 2016-03-26:
Hi there,

I have a related issue. I cannot plot topographies in the frequency domain to be save in any graphic format. I tried "figures, scalpmap SSEP series" which seems the only one allowing for this function as multiviewer no longer does that (compared to lets5).

Error in gui_mainfcn (line 96)
        feval(varargin{:});

Error in GLW_figure_scalpmap_SSEP_series (line 24)
    gui_mainfcn(gui_State, varargin{:});

Error in @(hObject,eventdata)GLW_figure_scalpmap_SSEP_series('headplot_btn_Callback',hObject,eventdata,guidata(hObject))

 
Error while evaluating uicontrol Callback

However in previous versions I was able to draw the topography.
Any suggestion?

Thanks and Happy Easter!

e

EVale, 2016-03-29:
I tried on other files and it seems that when applying the electrode coordinates both topo and headplot are displayed (both matlab 2010 and 2012)

amouraux, 2016-03-29:
Dear Elia,

Yes, the headplot function no longer requires to precompute the spline model. 
If electrode coordinates have been assigned, it should work. 

I know that the new multi viewer is missing the option to easily generate scalp maps other then in “live view”. Gan should put this function back soon, hopefully.

If I remember correctly, it is also possible to generate scalp maps / head plots in the “Explore ROI” function...

Best wishes,
André

EVale, 2016-03-29:
​Great Thanks!
I confirm that find ​"find peaks" function does work out the scalpmaps.

Best wishes,
e

EVale, 2016-04-08:
Hi Andre',

I tried that function and I get this output

[image]

It seems like I get a triangular representation than a spherical one.
Do I miss something in my folders?

A related issue is: in scalp map series if I ask for only 1 map I seem to obtain the max at the end of the scalpmap series rather than an average of the specified time period.
Do you confirm?

[image]

closed 2018-03-12 by amouraux