NOCIONS/letswave6 #64: View both datasets for math & stat operations?

JoanLS, 2016-02-02:
Hi André,

I have a minor GUI request. Would it be possible to see both dataset A & dataset B when doing mathematical operations as it was the case in LW5? Currently, I can only see one reference dataset (ie B). While it can be inferred that to the other one is A, I would prefer to be able to double-check this. This is also the case for paired t-tests. 

Thank you in advance!
Joan

amouraux, 2016-02-04:
Dear Joan,

This was chosen because dataset A can be more than one dataset…

An alternative would be to show a list of A datasets (i.e. all datasets that were selected when running the function minus the dataset selected as dataset B?)

André

JoanLS, 2016-02-05:
Hi André,

For me, the most intuitive way of using this would be to have a window with 3 columns.
One column contains all the selected datafiles from the LW window
The user can then select subsets from Col1 as Dataset A (second column) or Dataset B (third column). If there is only one dataset for A or B, the math operation is applying in reference to this single dataset. If there are equal numbers of datasets for A & B, the operation is done on a line-by-line basis (i.e. for each subject, subtract condition A from condition B). All the other cases would be illegal.

EVale, 2016-03-02:
Hi Andre',

I agree with Joan (or at least I would reintroduce the lets5 scheme). I also long ago was wondering if the maths can be done say between a merged file with several epochs and an avg file with 1 epoch (i.e. say you want to subtract group or single subj average from single subj/single trials).  In any case I love it!

amouraux, 2016-03-02:
Ok. 
Added to my todo list. 

I am also trying to find someone who could help me address all the open issues, as the number of users is increasing, and I am very short on time… Maybe someone in your lab with experience in Matlab would like to contribute?

Best wishes,
André

EVale, 2016-03-02:
I would love to!!!
Currently my lab is made by me and myself and a few undergrads that stop by
for short periods.
I have currently one engineer who applied for a Phd with me. If that works
then I will definitely ask her to contribute.

I may try myself but I will need some direction.

Best,
e

closed 2018-03-12 by amouraux