NOCIONS/letswave6 #68: ANOVA ON FFT data

EVale, 2016-03-02:
Hi Andre,

I tried to run an ANOVA on subtracted FFT snr files (3 levels within). 
I receive the following output

> C:1 Z:1 Y:1
> Warning: Matrix is singular to working precision. 
> > In anovaNxM>stats_glm_matrix (line 346)
>   In anovaNxM>stats_glm_anova (line 294)
>   In anovaNxM>callGLMForAnova (line 111)
>   In anovaNxM (line 102)
>   In ILW_custANOVA_parfor>makeF%1/F% (line 118)
>   In parallel_function (line 474)
>   In ILW_custANOVA_parfor (line 112)
>   In RLW_ANOVA (line 131)
>   In LW_ANOVA (line 82)
>   In letswave_gui>menu_callback (line 387) 
> Error using anovaNxM>stats_glm_matrix (line 346)
> Inner matrix dimensions must agree.
> 
> Error in anovaNxM>stats_glm_anova (line 294)
> R.red(p)=stats_glm_matrix(1,xred(p).p,y);
> 
> Error in anovaNxM>callGLMForAnova (line 111)
> R = stats_glm_anova(pname,xmat,data);
> 
> Error in anovaNxM (line 102)
> results=callGLMForAnova(data,pname,xmat,pnameErr,xmatErr,subjects);
> 
> Error in ILW_custANOVA_parfor>(parfor body) (line 118)
>                 result=anovaNxM(tpdata,tpsubjects,wtfactors,wtfactornames,btfactors,btfactornames);
> 
> Error in ILW_custANOVA_parfor (line 112)
>             parfor dy=1:header.datasize(5);
> 
> Error in RLW_ANOVA (line 131)
>     [out_dataset_pvalue,out_dataset_Fvalue]=ILW_custANOVA_parfor(datasets,within_levels,within_labels,between_levels,between_labels);
> 
> Error in LW_ANOVA (line 82)
>         [out_datasets(1),out_datasets(2),message_string,cluster_distribution]=RLW_ANOVA(datasets,configuration.parameters.factor_data,configuration.parameters.table_data,'enable_parallel',configuration.parameters.enable_parallel,'alpha',configuration.parameters.alpha,'permutation',configuration.parameters.permutation,'num_permutations',configuration.parameters.num_permutations,'cluster_statistic',configuration.parameters.cluster_statistic,'cluster_threshold',configuration.parameters.cluster_threshold);
        

> Error in letswave_gui>menu_callback (line 387)

However, when I run t-tests it goes smoothly and matrix singularity does not seem to be an issue.

amouraux, 2016-03-02:
Dear Elia,

I am fully aware of the problems with the ANOVA function. 

I am currently working on it. I decided to completely rewrite the function as the original third-party function that I used is buggy, and the authors does not really know how to fix it (anovaNxM). I am almost finished rewriting the low-level function. Then, I will need a few days to incorporate it in the GUI.

Note that you might be able to work around your problem by adding one between-subject factor with only one level. This will not change the results of your within-subject effects, but could solve the error message that you receive.

Best wishes,
André

amouraux, 2016-03-22:
ANOVA function has been completely rewritten. Hopefully, it should now work! (but it might be a bit slower than before)...

closed 2016-03-22 by amouraux