NOCIONS/letswave6 #70: Keep GUI window open after processing?

JoanLS, 2016-03-02:
Hi André,

Would it be possible to keep a GUI window open after pressing "process"? This would especially be useful when arranging or merging epochs because these steps cannot be scripted. I have tried to comment out the **close(handles.figure1);** line but that doesn't seem to work. 

Is there something else I should be changing?

Thanks in advance!

amouraux, 2016-03-21:
Dear Joan,

Unfortunately, this would not be possible with the current design. The closing of the figure is what triggers the LW function to process the data. That function needs to know when the parameters have been set... 

What would be possible would be to add a button making it possible to save/load a configuration. Would that help?

closed 2016-03-22 by amouraux

JoanLS, 2016-03-30:
Hi André,

Sorry for the late reply, I somehow couldn't log into Github...

The window-closing trigger doesn't seem to affect the plugin functions. Do you know if there is something different for them? Or how I could perhaps turn the functions I want to stay open into a plugin function? I have used the scripting function but I find that when I am trying several analysis parameters, it's easier to just select all the files and rerun the functions manually...

As for your suggestion, it would be extremely helful if the functions somehow automatically saved the last user parameters, especially if it's always the same values!