NOCIONS/letswave6 #71: Find peaks in waveforms

EVale, 2016-03-04:
[image]

It does not seem to display well the function.
Same it seems to apply to explore interval in multiviewer average.

[image]

In addition, it seems I get different values when I ask for max value in ROI generator compared with multiviewer. I will further check and let you know if that is a recursive phenomenon.

amouraux, 2016-03-22:
Display problem should now be fixed.
Can you confirm the problem with the max values obtained in ROI generator vs. multi viewer?

EVale, 2016-03-30:
Here is the display I get after the last update
[image]

Therefore, I could not extract the data.

EVale, 2016-07-05:
Hi!
I get the following error with the last update:

??? Error while evaluating uicontrol Callback

??? Undefined function or variable "table_data".

Error in ==> GLW_explore_ROI>process_btn_Callback at 465
uitable(h,'Data',table_data,'ColumnName',colheaders,'Units','normalized','Position', [0 0 1 1]);

Error in ==> gui_mainfcn at 96
        feval(varargin{:});

Error in ==> GLW_explore_ROI at 24
    gui_mainfcn(gui_State, varargin{:});

Error in ==> @(hObject,eventdata)GLW_explore_ROI('process_btn_Callback',hObject,eventdata,guidata(hObject))

[image]

closed 2016-07-05 by EVale