NOCIONS/letswave6 #84: objects out of the GUI

fedeAR24, 2016-04-18:
Hi André,

I noticed some of the objects in certain GUIs appear out of place, or they are not even present when running letswave in a PC. For example, the interface from CGLW_multi_viewer_average.fig. I solved it by manually finding the objects in the GUI using GUIDE and changing their position. I noticed that some of the objects have the "Units" Property set as "Normalized" instead of "Characters". That might create these problems when resizing the window in a PC. 
[http://se.mathworks.com/help/matlab/creating_guis/designing-for-cross-platform-compatibility.html#f8-1028670

amouraux, 2016-04-19:
Yes, the problem seems to be specific for panels.
I tried to fix these in most of the GUIs, but there are probably some remaining glitches.
Note that you can set some default viewing parameters using the Letswave File>GUI Options function.

fedeAR24, 2016-04-19:
Federico Arguissain :)

closed 2018-03-12 by amouraux