NOCIONS/letswave6 #85: Invalid Mex-File

hectordomorozco, 2016-04-19:
Hello, 

I am using the latest version of Letswave in Matlab R2016a (9.0.0.341360) in OS X El Capitan (10.11.4) in a MacBook Pro (2.9GHz Intel Core i5). I keep getting this mistake while importing EEGlab set files (and I get similar errors while trying to do anything):

Invalid MEX-file
'/Users/hectorOrozco/Documents/MATLAB/letswave6-master/external/fieldtrip/ft_getopt.mexmaci64':
dlopen(/Users/hectorOrozco/Documents/MATLAB/letswave6-master/external/fieldtrip/ft_getopt.mexmaci64, 6):
Symbol not found: _mxErrMsgTxt
  Referenced from:
  /Users/hectorOrozco/Documents/MATLAB/letswave6-master/external/fieldtrip/ft_getopt.mexmaci64
  Expected in: /Applications/MATLAB_R2016a.app/bin/maci64/libmx.dylib
 in /Users/hectorOrozco/Documents/MATLAB/letswave6-master/external/fieldtrip/ft_getopt.mexmaci64

Error in ft_read_data (line 58)
hdr           = ft_getopt(varargin, 'header');

Error in RLW_import_SET (line 26)
dat=ft_read_data(filename);

Error in LW_import_SET (line 86)
            [out_datasets(filepos).header out_datasets(filepos).data message_string]=RLW_import_SET(filename);

Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating Menu Callback

I've tried to uninstall and install again Letswave (multiple times now) to no avail. 

I really hope you guys can help me out. 

Thanks!

hughherald, 2016-04-20:
Dear hectordomorozco,
I met the same problem in the Win10 system a few days ago.
Now I update the mexw64 file, it should be OK.
Please check whether it works or not on your MAC system. 
It is really cool that you using the newest version of matlab. I notice that there are several new feature in it. 

Best,
Gan

amouraux, 2016-04-20:
Dear Gan,

I just updated the files myself! 
Oops!

Can you check whether it now works?
—
André Mouraux
Institute of Neuroscience (IONS)
Faculty of Medicine
University of Louvain
53, Avenue Mounier (B1.53.04)
B1200, Brussels, Belgium
Tel : +32.2.764.5447
Mobile : +32.499.670.486
http://www.nocions.org

hectordomorozco, 2016-04-20:
Hello, 

Thank you very much! It now works beautifully. 

Best, 
H

hughherald, 2016-04-21:
Dear Andre,
Your solution would be better. Previously, I have not read this part of codes entirely.
I think this problem may be caused by the version of Matlab updated.

Dear  Hectordomorozco,
It's glad that this problem is solved on your computer.

Best,
Gan

closed 2016-11-18 by hughherald