NOCIONS/letswave6 #94: Topography for a time range

hectordomorozco, 2016-05-30:
Hello, 

I have a quick question: is it possible to select a time range and plot a topography using the average power in said time range?

Thanks!

EVale, 2017-04-05:
Hi Andre', related to this: is it possible to add "mean" (or median) value for explore interval in multiviewer besides minimum and maximum? Is it enough to change GLW_multiview from line 984 to 1029 (I'm using lets 5 but I guess would be very similar for lets6 (?)

```
%launch figure;
topofigure=figure;
set(topofigure,'ToolBar','none');
%loop through selected files (will plot one scalp map per selected file, in separate columns)
for filepos=1:length(files);
    %loop through selected epochs (will plot one scalp map per selected epoch, in separate rows)
    for epochpos=1:length(epochs);
        %find maximum and minimum
        tp=squeeze(userdata(files(filepos)).data(epochs(epochpos),channels(1),index,dz,dy,dcursor1:dcursor2));
        [all_maxy,all_maxi]=max(tp);
        [all_miny,all_mini]=min(tp);
        if length(channels)>1
            for channelpos=2:length(channels);
                tp=squeeze(userdata(files(filepos)).data(epochs(epochpos),channels(channelpos),index,dz,dy,dcursor1:dcursor2));
                [maxy,maxi]=max(tp);
                [miny,mini]=min(tp);
                if maxy>all_maxy;
                    all_maxy=maxy;
                    all_maxi=maxi;
                end;
                if miny<all_miny;
                    all_miny=miny;
                    all_mini=mini;
                end;
            end;
        end;
        all_maxi=all_maxi+dcursor1-1;
        all_mini=all_mini+dcursor1-1;
        all_maxx=((all_maxi-1)*header.xstep)+header.xstart;
        all_minx=((all_mini-1)*header.xstep)+header.xstart;
        %graphpos
        graphpos=filepos+((epochpos-1)*length(files));
        %subplot
        tpaxis=subaxis(length(epochs),length(files),graphpos,'MarginLeft',0.01,'MarginRight',0.01,'MarginTop',0.05,'MarginBottom',0.01);
        if get(handles.scalppopup,'Value')==1
            dx=all_maxi;
        else
            dx=all_mini;
        end;
        tpaxis=LW_topoplot(header,userdata(files(filepos)).data,epochs(epochpos),index,dx,dy,dz,'maplimits',maplimits,'shading','interp','whitebk','on');
        [p,n,e]=fileparts(filestring{files(filepos)});
        st=[n,' - E:',epochstring{epochs(epochpos)}];
        st(findstr(st,'_'))=' ';
        title(gca,st);
    end;
end;
```

amouraux, 2017-04-11:
Dear Elia,

Actually, not that easy, because the Multiviewer function is very different in LW6. To allow for the online topographical maps, the function no longer allows to plot maps of min or max within an interval… :-/

André

closed 2018-03-12 by amouraux