NOCIONS/letswave6 #96: Generate table and graphs with ROI data error

GiuliaL, 2016-06-18:
Hi Andre,
I'm trying to generate a table with ROI data but I get the following error:

**Error: File: GLW_explore_ROI.m Line: 1296 Column: 1
Illegal use of reserved keyword "end".

Error in LW_explore_ROI (line 79)
        [a
        out_configuration]=GLW_explore_ROI('dummy',configuration,datasets);

Error in letswave_gui>menu_callback (line 329)
                configuration=eval(eval_st);
 
Error while evaluating uimenu Callback

Is it a bug or am I doing something wrong?**

Thanks!

amouraux, 2016-06-18:
Must be a bug. Will check later today or tomorrow morning...

amouraux, 2016-06-20:
Fixed.
Feature that is still missing is the ability to generate scalp maps of the selected ROI.

André

closed 2016-06-29 by amouraux