NOCIONS/letswave6 #100: Issue opening EEGLAB set files

karlinave, 2016-06-29:
Hello,

I am trying to open an EEGLAB set file using Letswave 6. When I go to File -> Import Signals -> EEGLAB set file, it recognizes my files, but when I choose them and then click process, I get the following error:

"Undefined function or variable "EEG".

Error in read_eeglabheader (line 45)
header.Fs          = EEG.srate;

Error in ft_read_header (line 570)
    hdr = read_eeglabheader(filename);

Error in ft_read_data (line 114)
  hdr = ft_read_header(filename, 'headerformat', headerformat);

Error in RLW_import_SET (line 26)
dat=ft_read_data(filename);

Error in LW_import_SET (line 86)
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_SET(filename);

Error in letswave_gui>menu_callback (line 387)

 
Error while evaluating Menu Callback"

I have not been able to figure out what the issue is, and I was wondering if you have any suggestions? Are there particular parameters for the EEGLAB set files that are required? I am importing EEGLAB set files that contain epoched EEG data. Any information or suggestions you have would be much appreciated!

Thanks,
Karli

amouraux, 2016-06-30:
Dear Karli,

Can you send me an example SET file so that I can try to figure out what is wrong?
The error is generated by the read_eeglabheader function, which is a fieldtrip function…

André

closed 2018-03-12 by amouraux

abhik9, 2021-03-04:
Hi,

I am trying to open an EEGLAB set file using Letswave 7. When I go to File -> Import Signals -> EEGLAB set file, it recognizes my files, but when I choose them and then click process, I get the similar error as above:

Undefined function or variable 'EEG'.

Error in read_eeglabheader (line 45)
header.Fs          = EEG.srate;

Error in ft_read_header (line 728)
    hdr = read_eeglabheader(filename);

Error in FLW_import_data.get_lwdata (line 243)
                hdr=ft_read_header(str);

Can you kindly look into matter and provide solution for it.

Encephalin, 2021-04-02:
Hi, was there ever a solution to this issue? I'm running into a similar error - 
[image]

amouraux, 2021-04-03:
Hello,

Could you send me an example SET file so that I can look into this issue?

Best wishes,
André

[Screen Shot 2021-04-02 at 2 29 08 PM]<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F10149605%2F113448288-89877e80-93c9-11eb-9c61-2fac739001ea.png&data=04%7C01%7Candre.mouraux%40uclouvain.be%7Cf18e04ed6354424f920c08d8f60f0341%7C7ab090d4fa2e4ecfbc7c4127b4d582ec%7C0%7C0%7C637529891634187049%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=8SGfHNxYxr5cYIiG4%2FM7KJ1lilGZcNxQIqZd2wpFQJ0%3D&reserved=0>

jimark13, 2021-06-05:
Hi, 
Has there been a solution, I am running across the same issue?

BEst, 
Jasper

BenedettaCecconi, 2024-02-18:
Hello,

I'm having the same problem as [karlinave](https://github.com/karlinave). 

I see that this issue has been closed but I can't see the solution in the thread.. Can someone tell me what was the solution?

Thanks,

sespinozae, 2024-03-05:
If you preprocess your data in a newer version of eeglab, you need to replace the version of the read_eeglabheader.m with this version (https://github.com/fieldtrip/fieldtrip/blob/master/fileio/private/read_eeglabheader.m)

Hope it helps!