NOCIONS/letswave6 #102: Not possible to build figure map series with plugin in letswave 6-master

sofievettori, 2016-06-30:
Hi,

It is not possible to build figure map series with the plugin. Neither 2D or 3D maps are possible. This is the error message in matlab: 
Undefined function 'exist' for input arguments of type 'struct'.

Error in headplot (line 470)
  if ~exist(spline_file)

Error in LW_headplot (line 22)
h=headplot(vector2,header,varargin{:});

Error in GLW_figure_mapseries_extended>processbutton_Callback (line 356)
            LW_headplot(header,dataCell{rowpos},epochpos,index,dxi(i),dy,dz,'maplimits',[cmin
            cmax],'electrodes',electrodeOn);

Error in gui_mainfcn (line 96)
        feval(varargin{:});

Error in GLW_figure_mapseries_extended (line 23)
    gui_mainfcn(gui_State, varargin{:});

Error in
@(hObject,eventdata)GLW_figure_mapseries_extended('processbutton_Callback',hObject,eventdata,guidata(hObject))

 
Error while evaluating uicontrol Callback

Could you please help me with this issue?
Thank you very much!
Sofie

amouraux, 2016-07-07:
Should be fixed now.

André

closed 2018-03-12 by amouraux

Evgevaie, 2020-06-05:
Hi,

Unfortunately, this error still persists. For 3D head maps only (for 2D works fine). 
Latest version of Letswave6; Matlab2012b and 2018b; Windows 10.

Could you fix it, please?..

Best,
Evgenia