NOCIONS/letswave6 #103: Failure starting parallel pool with ANOVA

EVale, 2016-07-05:
Not sure whether my be  alocal issue...

Error: File: ILW_custANOVA_parfor.m Line: 200 Column: 25
The variable outdataset_pvalue in a parfor cannot be classified.
See Parallel for Loops in MATLAB, "Overview".

Error in RLW_ANOVA (line 131)
    [out_dataset_pvalue,out_dataset_Fvalue]=ILW_custANOVA_parfor(datasets,within_levels,within_labels,between_levels,between_labels);

Error in LW_ANOVA (line 82)
        [out_datasets(1),out_datasets(2),message_string,cluster_distribution]=RLW_ANOVA(datasets,configuration.parameters.factor_data,configuration.parameters.table_data,'enable_parallel',configuration.parameters.enable_parallel,'alpha',configuration.parameters.alpha,'permutation',configuration.parameters.permutation,'num_permutations',configuration.parameters.num_permutations,'cluster_statistic',configuration.parameters.cluster_statistic,'cluster_threshold',configuration.parameters.cluster_threshold);
        
Error in letswave_gui>menu_callback (line 387)

 
Error while evaluating Menu Callback

Error using parpool (line 94)
Failed to start a parallel pool. (For information in addition to the causing error, validate the profile 'local' in the Cluster Profile Manager.)

EVale, 2016-08-30:
I replicated this error on the same machine (dell optiplex9020 win 10) after updating the software.

Error: File: ILW_custANOVA_parfor.m Line: 200 Column: 25
The variable outdataset_pvalue in a parfor cannot be classified.
See Parallel for Loops in MATLAB, "Overview".

Error in RLW_ANOVA (line 131)
    [out_dataset_pvalue,out_dataset_Fvalue]=ILW_custANOVA_parfor(datasets,within_levels,within_labels,between_levels,between_labels);

Error in LW_ANOVA (line 82)
        [out_datasets(1),out_datasets(2),message_string,cluster_distribution]=RLW_ANOVA(datasets,configuration.parameters.factor_data,configuration.parameters.table_data,'enable_parallel',configuration.parameters.enable_parallel,'alpha',configuration.parameters.alpha,'permutation',configuration.parameters.permutation,'num_permutations',configuration.parameters.num_permutations,'cluster_statistic',configuration.parameters.cluster_statistic,'cluster_threshold',configuration.parameters.cluster_threshold);
        
Error in letswave_gui>menu_callback (line 387)

 
Error while evaluating Menu Callback

amouraux, 2017-01-16:
Can you check if this is now fixed?

closed 2017-01-16 by amouraux

EVale, 2017-02-06:
Hi Andre,

I updated to the last version and tried it out but I cannot really
understand if it is working.
Once I launch it it stays on the screen like this

Starting parallel pool (parpool) using the 'local' profile ... >>

I'm not sure of how long I am supposed to be waiting but after 10 min was
still like that...

Best,

e

amouraux, 2017-02-17:
Dear Elia,

There is, apparently, a problem in the parallel pool version of the function.
As we are now reaching the stage of finalising Letswave7 with Gan, and as the new version has a completely rewritten ANOVA function which is about 100x faster than the current function…
We still need to transfer some missing functions, but we do hope to release it within a few weeks…

Best wishes,
André

EVale, 2017-02-17:
Great news!!!!

looking forward to try the new version out!

e

amouraux, 2017-02-17:
Dear Elia,

Actually, you can already try it, I it is developed on github. Just, expect some bugs (not many) and missing features (many).

https://github.com/NOCIONS/letswave7

Note that you can run Letswave7 along Letswave6 without any problem.

Best wishes,
André

EVale, 2017-02-17:
> Tel : +32.2.764.5447 <+32%202%20764%2054%2047>

Ehi,

I like it!
does the co-running future applies also to Lets5?
I received this error while launching multiviewer in Lets5 with Lets7
opened.
it is an issue with the saved path in matlab.

Error using load
Unable to read file
'C:\Users\evalent\Documents\MATLAB\letswave7-master\settings\multiview_settings.mat'.
No such file or directory.

Error in GLW_multiview_GLW_multiview_OpeningFcn (line 336)
load(localtarget);

Error in gui_mainfcn (line 220)
    feval(gui_State.gui_OpeningFcn, gui_hFigure, [], guidata(gui_hFigure),
varargin{:});

Error in GLW_multiview (line 34)
    gui_mainfcn(gui_State, varargin{:});

Error in letswave_gui>proc_multiview_Callback (line 1252)

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in letswave_gui (line 32)

Error in
@(hObject,eventdata)letswave_gui('proc_multiview_Callback',hObject,eventdata,guidata(hObject))

Error while evaluating Menu Callback

amouraux, 2017-02-17:
Actually,

What I said was a bit over-enthusiastic. You can have your path saved to LW6 and LW7, and choose which version you want to run, but once it is running, you will get into trouble if you use both at the same time, because several functions have the same name…. Sorry!
André