NOCIONS/letswave6 #106: Error with files

KingFunk, 2016-08-26:
I'm getting the following errors on all my EEG files after preprocessing:

Preprocess steps taken:
Frequency Filter
Baseline Correction
Channel labeling and deleting unused channels
Epoch sorting
Merging

When I try to open the files past the baseline correction step, I get the following error and it does not open.  Any ideas what could cause this?

Warning: Imaginary parts of complex X and/or Y arguments ignored 
> In GLW_simpleview>updategraph at 489
  In GLW_simpleview>GLW_simpleview_OpeningFcn at 115
  In gui_mainfcn at 221
  In GLW_simpleview at 33
  In letswave_gui>filebox_Callback at 106
  In gui_mainfcn at 96
  In letswave_gui at 32
  In @(hObject,eventdata)letswave_gui('filebox_Callback',hObject,eventdata,guidata(hObject))

amouraux, 2016-08-26:
Hello,

What function did you use to filter? An FFT filter?
It seems that your data is composed of complex numbers instead of real numbers. 
This could have occurred if there was a mistake when setting the parameters of an FFT filter.
Can you check whether you are able to visualise the filtered data?
Can you provide me with the parameters of the FFT filter?
If you are unsure, you can send me as attachment the header file, i.e. the LW6 file as it contains a history of all the different processing steps.

Best wishes,
André

KingFunk, 2016-08-26:
Analysis was performed in letswave5.   A FFT filter was used with a
band-pass filter at default settings.

I can also visual the filtered data and have attached the header file for
the filter and the last processing step done.

Thanks for the help!

-Bob

KingFunk, 2016-08-26:
[data file headers.zip](https://github.com/NOCIONS/letswave6/files/439389/data.file.headers.zip)

closed 2017-01-16 by amouraux