NOCIONS/letswave6 #109: compute area under the curve

giulia92222, 2016-09-29:
Dear all,

I'm using letswave6 with EEG files and I would like to know how I can compute the area under the curve, maybe is it something related to the section "multiviewer"? I found the function "find peaks" (min and max) but I would need more informations about computing areas...

Thank you so much

amouraux, 2016-10-01:
You should be able to do this in the “Explore ROI” function.
Can you get back to me if this does not work?

Kind regards,
André

giulia92222, 2016-10-08:
Dear Prof. Mouraux

thank you very much for your answer.

Unfortunately I don't find the function. I found the function "generate table and graphs 
with ROI" under "View" menu, but there is no the "area under the curve information".
Am I wrong?

In LW4 I obtained the Area under the Curve right clicking on the wa4 file:
"explore - waves, extract information"

Thanks again,

Giulia Guerra

valebrown, 2018-02-28:
Dear Giulia, 
I have your same problem. Did you understand where to find the "Explore ROI" function?

Thank you, 
Valentina

amouraux, 2018-03-11:
Sorry. The function is actually called "Generate table and graphs with ROI data" in the View menu.
I have now added the AUC calculation, which will appear as an additional column when you click "Generate Table".

closed 2018-03-11 by amouraux

valebrown, 2018-03-12:
Dear Prof. Mouraux, 
thank you very much, it perfectly works.

Kind regards,