NOCIONS/letswave6 #110: The unnecessary use of the neural network toolbox

dannem, 2016-10-03:
Hello,
I am trying to promote the use of letswave at my university (University of Toronto).
However, there is a difficulty with using this toolbox since it relies on neural networks Matlab toolbox for performing unrelated (to the neural networks) tasks, such as view functions.
Please find the error message attached below. Would it be possible to exchange these nnet functions with something which is available in default Matlab versions? 

Thank you and best regards,
Dan

dist is not included in your installed products. These products offer 'dist':
  Neural Network Toolbox

Error in CLW_make_spl (line 45)
[~,I]=sort(dist([Xe Ye Ze],spherePOS'),2);

Error in CGLW_multi_viewer/init_parameter (line 33)
            datasets_header(k).header=CLW_make_spl(datasets_header(k).header);

Error in CGLW_multi_viewer/CGLW_my_view_OpeningFcn (line 477)
        init_parameter();

Error in CGLW_multi_viewer (line 15)
CGLW_my_view_OpeningFcn;

Error in letswave_gui>view_files (line 951)
    CGLW_multi_viewer(cb,inputfiles,send_update_status(handles));

Error in letswave_gui>view_btn_Callback (line 983)
view_files(handles);

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in letswave_gui (line 34)
    gui_mainfcn(gui_State, varargin{:});

Error in matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)letswave_gui('view_btn_Callback',hObject,eventdata,guidata(hObject))
 
Error while evaluating UIControl Callback

hughherald, 2016-11-18:
Dear dannem,
I'm Gan Huang, I try to fix the issue you proposed. I have not test it yet, because it would be time consuming for setup the environment. Can you update the Letwave6 to see whether the problem is fixed or not.

dannem, 2016-12-02:
Thanks Gan, 
It seems to be working now. If there are more issues, I will let you know.
We like your software. 
Cheers,
Dan

hughherald, 2016-12-28:
Thanks Dan,

Merry X'mas,
Gan

closed 2016-12-28 by hughherald