NOCIONS/letswave6 #112: Import VHDR

Manolo68, 2016-10-17:
Hi,
When I try do import signal and click on "import brainvision vhdr files", the window opens but with a strange format and I can't choose the folder where my files are. Even if I try to maximize the window or else, I cannot do nothing and even the buttoon "process" is almost invisible. It looks like a graphic bug but could not figure it out. Even more annoying, it worked perfectly the last time I used letswave 6 so can't really understand what happened.
Anyone had the same problem or knows how to fix it ? 

thanks for your help,
best.
Manu

rachelrhs20, 2016-11-29:
Hi, 
Same problem here, would be grateful for a fix if anyone knows of one.
Thank you for your help.
Rachel

MegaCurry, 2017-01-16:
Hi,
I am also faced with the same problem with you guys. Not only when "import vhdr files", but also in "import ASCII". It seems a GUI problem. I searched the net but found it's hard to change the options by scripts.

Thank you for your help
Moqian

amouraux, 2017-01-16:
This should now be fixed after you update.

closed 2017-01-16 by amouraux