NOCIONS/letswave6 #118: Can't import BrainVision VHDR files

rachelrhs20, 2016-11-29:
When I do File>Import signals>Import BrainVision VHDR files, the dialog box for file selection pops up at the wrong size. 
I can see only a corner of the box, it will not resize, and therefore it is not possible to select files for import.
Any solutions gratefully received.
Thanks for your help.
Rachel

amouraux, 2016-11-29:
Hello,

I should be able to fix this easily, probably tomorrow when I have access to a Windows machine.
It is a Mac/Windows problem when we edit the GUIs.
I thought that I had solved this previously, can you confirm that you have the latest version?

Best wishes,
André

rachelrhs20, 2016-11-29:
Hi Andre

Thanks very much for your prompt response.

I downloaded the software today. The zip file is NOCIONS-letswave6-6.0-237-g4be7eca.zip and I got it from http://nocions.github.io/letswave6/

Now I see that both www.nocions.org/letswave<http://www.nocions.org/letswave> and https://github.com/NOCIONS/letswave6/wiki/Download-and-setup link to a file called master.zip

Shall I try that one before putting you to the trouble of (re-)fixing the problem?

Thanks again

Rachel

rachelrhs20, 2016-11-29:
Hello again Andre

The problem is still there when I use the letswave6-master.zip file, unfortunately.

sorry! and thank you

Rachel

amouraux, 2016-12-01:
I just fixed the visualisation glitch in all import functions. It should work fine after updating. Seems to be related to a weird behaviour of Matlab GUIs across platforms. Please tell me if you find any other function with such a visualisation problem.

closed 2016-12-01 by amouraux

rachelrhs20, 2016-12-01:
Hi Andre
Thank you so much! Really appreciate this. I will let you know if I come across any other issues.
Best wishes
Rachel