NOCIONS/letswave6 #121: Reject epochs (visual inspection)

HugotenDam, 2017-02-13:
Hi dear people from letswave, 

When i try to inspect my data visually the window won't open in a normal way, not allowing me to select channels or properly inspect the data (see picture for how the window looks). I can't adjust the window by scrolling or sliding the scroll bar. 

[image]

amouraux, 2017-02-15:
Hello.
Should be fixed now after updating.
This is really a weird bug, related to some compatibility issue between Mac and Windows.

Best wishes,
André

[image]<https://cloud.githubusercontent.com/assets/25753279/22900728/ae3c1442-f22f-11e6-8229-68e9bc933543.png>

HugotenDam, 2017-02-22:
> Tel : +32.2.764.5447 <+32%202%20764%2054%2047>

Hi Andre,

You are right. I downloaded the package again and now it works. However, i
can't manage to fill in the amplitude criterion upon visual inspection
(see picture). Do you know (once again) why?

Kind regards,

Hugo

[image: Inline afbeelding 1]

Met vriendelijke groet,

Hugo ten Dam

amouraux, 2017-03-07:
I cannot view the attached picture, and I am able to fill in the amplitude criterion on my own setup. Could you send the picture again?

Best wishes,
André

HugotenDam, 2017-03-07:
Hi André,

Strange: I can open and view it.
Now I attached the image.

Kind regards,

Hugo ten Dam

Met vriendelijke groet,

Hugo ten Dam

closed 2018-03-11 by amouraux