NOCIONS/letswave6 #126: ERROR in Import BDF

ValeTheIstrice, 2017-05-08:
Dears,

when I try to import a .BDF  eeg raw data it gives me this error:

"??? Error using ==> bitcmp
Inputs must be non-negative integers.

Error in ==> ft_read_event at 336
      sdata(bit24i) = bitcmp(abs(sdata(bit24i))-1,24);

Error in ==> RLW_import_BDF at 32
trg=ft_read_event(filename);

Error in ==> LW_import_BDF at 86
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_BDF(filename);

Error in ==> letswave_gui>menu_callback at 387
                [configuration,datasets]=eval(eval_st);
 
??? Error while evaluating uimenu Callback"

Do you know on what could it depends?

amouraux, 2017-05-08:
Error is probably due to the fact that Matlab changed the syntax for the bitcmp function.
I now updated the functions to include some of the bug fixes in that fieldtrip function.
Can you update your Letswave version and try again?

If it still does not work, I guess that you will have to send me a sample dataset…

Best wishes,
André

ValeTheIstrice, 2017-05-08:
> Tel : +32.2.764.5447 <+32%202%20764%2054%2047>

Actually, I also have a problem in the updating :

??? Error using ==> checkfilename>validateFilename at 184
Function UNZIP was unable to find file 'C:\Program
Files\MATLAB\R2010a\toolbox\letswave6-master\letswave6.zip'.

Error in ==> checkfilename at 50
[fullfilename, fid] = validateFilename( ...

Error in ==> parseUnArchiveInputs at 87
[archiveFilename, url] = checkfilename(archiveFilename, validExtensions, ...

Error in ==> unzip at 58
[zipFilename, outputDir, url, urlFilename] = parseUnArchiveInputs( ...

Error in ==> CGLW_update>process_btn_Callback at 98
filenames=unzip(filename,p);

Error in ==> gui_mainfcn at 96
        feval(varargin{:});

Error in ==> CGLW_update at 24
    gui_mainfcn(gui_State, varargin{:});

Error in ==>
@(hObject,eventdata)CGLW_update('process_btn_Callback',hObject,eventdata,guidata(hObject))

??? Error while evaluating uicontrol Callback

Is the first time that I update the version instead to download a new one,
maybe I did a mistake? I went into "File" in the menu, then I selected
"Check for updates...", It checked but when I click on Update it gives me
the error.

amouraux, 2017-05-08:
Strange.
But I do notice that you are running a very old version of Matlab, 2010a.

You can update manually by downloading the ZIP file and unzipping it in the letswave6-master folder, such as to overwrite all the old files.

Best wishes,
André

ValeTheIstrice, 2017-05-09:
> valentina-nicolardi#anchor

Thanks! I have updated manually but still this error when I try to import
BDF:

??? Error while evaluating uicontrol Callback

??? Error using ==> bitcmp
Inputs must be non-negative integers.

Error in ==> ft_read_event at 336
      sdata(bit24i) = bitcmp(abs(sdata(bit24i))-1,24);

Error in ==> RLW_import_BDF at 32
trg=ft_read_event(filename);

Error in ==> LW_import_BDF at 86
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_BDF(filename);

Error in ==> letswave_gui>menu_callback at 362
                [configuration,datasets]=eval(eval_st);

??? Error while evaluating uimenu Callback

Maybe I should try with Letswave 5, due to my Matlab version?

amouraux, 2018-03-11:
I updated the field trip functions. Do you still have the issue?

nihanalp, 2019-06-19:
I also face a problem when I try to import a bdf file. 

nvalid MEX-file '/Users/nihanalp/Dropbox/mat601nihan/letswave5/external/fieldtrip/ft_getopt.mexmaci64':
dlopen(/Users/nihanalp/Dropbox/mat601nihan/letswave5/external/fieldtrip/ft_getopt.mexmaci64, 6): Library not
loaded: @loader_path/libmex.dylib
  Referenced from: /Users/nihanalp/Dropbox/mat601nihan/letswave5/external/fieldtrip/ft_getopt.mexmaci64
  Reason: image not found.

Error in ft_read_data (line 58)
hdr           = ft_getopt(varargin, 'header');

Error in LW_importBDF (line 27)
dat=ft_read_data(filename);

Error in GLW_importBDF>pushbutton4_Callback (line 140)
    [header,data]=LW_importBDF(filenames{filepos});

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in GLW_importBDF (line 33)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)GLW_importBDF('pushbutton4_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.

Could you please help me with this?