NOCIONS/letswave6 #127: electrode location

DT83, 2017-05-15:
Is it possible that the function electrode location does not function properly when channel locations others than the standard availble one are used? 

I have imported and loaded the Biosemi_128 (worked in EEGlab), it seems to work (no error), but it is not recognized afterwards (when I tried to plot the scalptopography)

amouraux, 2018-03-11:
How did you import channel locations in Letswave?

DT83, 2018-03-15:
That's easy to reply: i used the .ela file (129 channels). I can send it to
you

 I also figured out why the number of channels did not correspond (it had to do with the fact that eeglab, which i previously used to import eliminates the refernce channel if not specified otherwise).
Thanks for your help!

Inviato da iPhone