NOCIONS/letswave6 #128: Bug in Map Multiviewer figure

Manolo68, 2017-05-23:
Dear,
I noticed that with the last update of LW6, I have a bug when I open the Map Multiviewer of an average or grand average (X and Y options are not visible anymore, the window of the MM is half cut)
Am I the only one ? Could you please fix this issue ? 

Thank you very much for your work !
Best,

manu

mgrabo, 2017-08-03:
Hi Manu, 

I have the same problem. I am working with windows 10 and Matlab 2016b. It would be great if this issue could be fixed. 
Thank you very much for your work! 

Kind regards,
Martyna

GiuliaL, 2018-01-30:
Hi, 
I have the same issue...

Giulia

amouraux, 2018-03-11:
Very annoying Matlab bug causing Mac <> Windows display issues. Should be fixed now.

closed 2018-03-11 by amouraux