NOCIONS/letswave6 #130: Letswave 6. Grand Average (weighted)

sabaristei, 2017-05-29:
Dear André, 
I used Grand Average on SNR data (subtraction). It is usually a smooth operation. However, this time the function reports an error:
>Array dimensions must match for binary array op.
>Error in RLW_grand_average (line 60)
>    out_data=out_data+(data*dataset_weights(setpos));

>Error in LW_grand_average (line 76)
>[out_datasets(1).header,out_datasets(1).data,message_string]=RLW_grand_average(datasets,'dataset_weigh>ts',configuration.parameters.dataset_weights);
        
>Error in letswave_gui>menu_callback (line 387)
>Error while evaluating Menu Callback.

Arrays have all 6 dimensions, the only dimension with different values is the X-dimension. I used the values of X for each dataset to set the weights, but nothing changed. Could you suggest a way to solve the issue?
Thank you in advance.

Best,
Sabrina

amouraux, 2017-05-29:
If you get this error, it means that the different datasets do not have the same number of data points per epoch, or the same number of channels, index, etc…
Can you double check?

Best wishes,
André

sabaristei, 2017-05-29:
...checked again: indices, channels, epochs are identical, as they are Y and Z size, but different data points (both XStep and X Srate vary across datasets). Is there a way around it?
Thank you for helping out.

Sabrina

amouraux, 2017-05-29:
Could you resample them (using the resample function) so that they have matching xsize and xstep?

André

sabaristei, 2017-05-29:
...many thanks. It works!

Best wishes,
Sabrina

closed 2018-03-11 by amouraux