NOCIONS/letswave6 #137: Errors importing .avg EEProbe files

lkcirelli, 2017-12-20:
Hello, 

I am using Matlab_R2016b (9.1.0.441655) in OS X Yosemite (10.10.5) in a macbook air (1.3GHz Intel Core i5). I am trying to import .avg files from EEProbe, something that I often did with no issues using letswave in the past. I am getting different errors with Letswave 5 and Letswave 6. 

With Letswave 6: 
Undefined function or variable 'removefields'.

Error in ft_read_header (line 711)
    hdr = removefields(hdr, {'data', 'variance'});

Error in ft_read_data (line 200)
  hdr = ft_read_header(filename, 'headerformat', headerformat, 'chanindx', chanindx,
  'checkmaxfilter', checkmaxfilter);

Error in RLW_import_CNT (line 44)
dat=ft_read_data(filename);

Error in LW_import_CNT (line 87)
            [tp_datasets
            message_string]=RLW_import_CNT(filename,'split_blocks',configuration.parameters.split_blocks);
            
Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating Menu Callback

With Letswave 5:

Invalid MEX-file '/Users/lauracirelli/Dropbox/letswave5/external/fieldtrip/ft_getopt.mexmaci64':
dlopen(/Users/lauracirelli/Dropbox/letswave5/external/fieldtrip/ft_getopt.mexmaci64, 6): Symbol
not found: _mxErrMsgTxt
  Referenced from: /Users/lauracirelli/Dropbox/letswave5/external/fieldtrip/ft_getopt.mexmaci64
  Expected in: /Applications/MATLAB_R2016b.app/bin/maci64/libmx.dylib
 in /Users/lauracirelli/Dropbox/letswave5/external/fieldtrip/ft_getopt.mexmaci64.

Error in ft_read_data (line 58)
hdr           = ft_getopt(varargin, 'header');

Error in LW_importCNT (line 27)
dat=ft_read_data(filename);

Error in GLW_importCNT>pushbutton4_Callback (line 144)
    [header,data]=LW_importCNT(filenames{filepos});

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in GLW_importCNT (line 33)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)GLW_importCNT('pushbutton4_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback

Any help with this error on either version of Letswave would be greatly appreciated. 

Thanks!
Laura

amouraux, 2017-12-22:
Seems to be a bug in fieldtrip.
I just added the missing function.
Can you update LW6 and try again?

Best wishes,
André

lkcirelli, 2017-12-22:
Hi André, 

Thank you for looking into this bug! The files now load into Letswave without any error messages, but there is something wrong with the resulting files. They certainly don't look like EEG waveforms (see screenshot below - similar on all channels). Any suggestions? I've tried with a few different .avg EEProbe files that I have imported in the past with no issue. 

Cheers,
Laura

[image]

lkcirelli, 2018-01-24:
Hello André, 
Sorry for posting so close to the holidays before! Any thoughts on the errors I'm having?
Cheers,
Laura

amouraux, 2018-03-11:
Is the issue fixed? Else, can you send me a sample dataset? Thanks, André.

lkcirelli, 2018-03-13:
Hello André, 
Thank you for following up. (sorry, I’m new to GitHub and replied there instead of on this email chain last Dec). 
The files now load into Letswave without any error messages, but there is something wrong with the resulting files. They certainly don't look like EEG waveforms (see screenshot below - similar on all channels). Any suggestions? I've tried with a few different .avg EEProbe files that I have imported in the past with no issue.
Here is a dropbox link to one of my .avg files: https://www.dropbox.com/s/691zdqoyxlk45ik/sampleData_Cirelli.avr?dl=0
Thanks very much,
Laura