NOCIONS/letswave6 #138: FFT multi-notch Filter_no harmonics?

Razieh-Alemi, 2018-01-19:
Hello,
I would like to apply FFT multi-notch filter with harmonics on my dataset, but I am not able to find any part related to selecting number of harmonics!
I am working on Letswave 6.
Would you please tell me where I can find that option. Thanks for your consideration.

Best
Razieh

dmulders, 2018-01-19:
Hello,
If you want to keep the fundamental frequency and some harmonics (let's say f1, 2*f1, 3*f1...), you can just indicate all these frequencies in the field "notch frequencies" of the "FFTmulti-notch filter" dialog box. 

Best,
Dounia

Razieh-Alemi, 2018-01-23:
Thanks Dounica for your help. Actually I want to apply Notch FFT filter, 60Hz, (with10 harmonics). So would you please let me know what is the best way to do this? Cause on wiki page I found the part related to this, but it is not clear on the software.
Thanks for your consideration.
Best,
Razieh

dmulders, 2018-01-23:
You can hence specify the values [60   120   180   240   300   360   420   480   540   600  660] in the field "notch frequencies". I guess that you want to remove these frequencies (and not select them), so don't forget to untick the line "Inverse notch filter" when you apply the filter. Otherwise, do I misunderstand your question?

Best,
Dounia

Razieh-Alemi, 2018-01-23:
That is great! Thanks a lot!
Best
Razieh

closed 2018-03-11 by amouraux