NOCIONS/letswave6 #139: Number of channels not corresponding from original file to Letswave

DT83, 2018-01-25:
Hi, 

when importing files from the Geodesic system (129 channels with Cz as channel 129 and reference) I end up having only 128 channels visible in the letswave file and no possibility of seeing Cz (electrode 129). 

Does this depend on an exporting or importing issue? I am currently using the .sfp file

Best

Diana

amouraux, 2018-03-11:
Could you send me a sample file to test this out?

amouraux, 2018-03-11:
I updated the EGI import function to include another implementation using the java toolbox provided by EGI. Both implementations actually use the fieldtrip EGI import function. Can you check whether you also have the issue when you use the other implementation method?