NOCIONS/letswave6 #143: cluster based permutations Anova with different letswave6

valebrown, 2018-03-13:
Good evening, 
I had a problem with the cluster-based permutations Anova. I usually run the 250 permutations Anova with the first letswave release (lestwave 6-6.0, https://github.com/NOCIONS/letswave6/releases) and it takes about 2-2.5 hours. However, when I run the same 250 permutations Anova, with the same data, on the same pc, with the same Matlab version (2015) with the latest Letswave version (letswave6-master) it takes about 41 hours!
The results of the two Anova appear to be the same, but I would kindly ask why there is this difference between the two letswave versions. Is there a reason? 

Thank you, 
Valentina

amouraux, 2018-03-13:
Hello,

We corrected some bugs which fixed some important issues, but resulted in a much slower execution.
We have further improved the function in LW7, resulting in a marked improvement (approximately 10x faster than the original version).
Because we are close to finalising LW7, and because it would require quite a lot of work to transfer this to LW6, you may want to try the beta version of LW7, which is available for download on GitHub…

Best wishes,
André

valebrown, 2018-03-13:
Dear prof Mouraux, 
thank you! I'll certainly try with LW7.

Best regards,

cuic0808, 2020-09-28:
Hello all, I want to ask a question about letswave7, there was no permutation settings in multiple factor(2x2 in my desin) ANOVA in letswave7, which version can I use to complete permutataion ANOVA.

Thank you very much!

Best regards.