NOCIONS/letswave6 #144: Not possible to build 3D plots (SS-EP)

fchetail, 2018-03-22:
Hello,
I've two issues with topographies:

-  If I use multiviewer to have a quick look at signal at a given base frequency, it's more or less ok in 2D (left part is missing, while electrodes are present, I don't understand why), but if I want to see the signal in 3D, I just have the head with the electrodes. The same happens if I use View/Figures/Scalpmap SSEP series figure creator (though that time 2D is fine).

- The plugin 'Figure map series (extended)' doesn't seem to work. I read in a previous that it was fixed, but here is the error:

Error using exist
The first input to exist must be a string scalar or character vector.

Error in headplot (line 470)
  if ~exist(spline_file)

Error in LW_headplot (line 22)
h=headplot(vector2,header,varargin{:});

Error in GLW_figure_mapseries_extended>processbutton_Callback (line 356)
            LW_headplot(header,dataCell{rowpos},epochpos,index,dxi(i),dy,dz,'maplimits',[cmin
            cmax],'electrodes',electrodeOn);

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in GLW_figure_mapseries_extended (line 23)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)GLW_figure_mapseries_extended('processbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.

Many thanks in advance for your answers
Best
Fabienne

[image]

amouraux, 2018-03-26:
Can you update to the latest version and check whether you still have the error?
From the line numbers, it seems that you do not have the latest version of the files.
If the problem persists, can you send me an example dataset to understand why you are having this problem?

Best wishes,
André

[3d_topo]<https://user-images.githubusercontent.com/37670073/37790486-fc64b9a8-2e06-11e8-8136-71a2f8fcbaf2.png>

fchetail, 2018-03-27:
André,

Thanks for your swift reply. I finally noticed that the error occurred only for one type of files (SNR). This was due to the fact that a channel had only NaN values (channel used for off-line re-referencing, thus equal to 0 before SNR transform). When this channel is removed, everything works!
Do you want me to update my post on forum?

Best
Fabienne

Razieh-Alemi, 2018-04-04:
Hello,
That happened to me, too. Reloading electrode location file fixed it (cause I did preprocessing with EEGLAB).
Best,
Rzaieh

fchetail, 2018-04-04:
Ok, thanks Rzaieh!

closed 2018-04-05 by fchetail