NOCIONS/letswave6 #145: Re-referencing with linked mastoids

fchetail, 2018-04-05:
Hello,
During recording, I used the right mastoid as online reference. Now, I want to re-reference with the left mastoid to avoid any hemispheric bias (linked mastoid re-referencing). To do so, the mathematical operation consists in subtracting the signal of the left mastoid divided by 2 to all the other electrodes. How can I do that in LW6? So far, it seems to me that I can only subtract the signal of the left mastoid to all the other electrodes, which keeps the bias. It's important to mention that the channel corresponding to the online reference (= 0) is not imported during upload so that I cannot reference with the mean of the two electrodes (which would have enabled me to create the linked mastoid...).  
Many thanks in advance,
Fabienne

dmulders, 2018-04-05:
Hello Fabienne,

You can "create" new channels with half the amplitude of your original signals with the following procedure:
1) Postprocess --> Math --> Math operation using a constant --> Divide (A/constant) (choosing constant = 2)
2) Edit --> Electrodes --> Edit electrode labels: you can rename the electrode of your left mastoid to be sure to not confound it with the left mastoid before division by 2 (this step is not mandatory)
3) Preprocess --> Arrange signals --> Merge channels of multiple data sets
Now the merged data set contains both the original channels and these same channels but with an amplitude divided by 2. You can hence re-reference as usual, and then remove all the channels that you have created for the re-referencing (almost all the channels can also already be removed between steps (2) and (3) above).
I know this is quite long for a simple operation so we will try to ease this procedure in a future release!

Best regards,

Dounia

fchetail, 2018-04-06:
Thanks very much Dounia!
It works fine, although as you said, this is quite long for such a simple operation :-)
Best
Fabienne

closed 2018-04-06 by fchetail