NOCIONS/letswave6 #146: Raw Matlab script with LW functions

fchetail, 2018-04-20:
Hello,

I was wondering whether you guys developed something to help users who want to script LW functions (real scripting, not using the function 'script' of the LW GUI).
In the toolbox EEgLab, each time a function is used via the GUI, it is possible to generate the corresponding Matlab code with all the desired parameters ('eegh' typed into the console). This enabled me to very easily and rapidly build an entire Matlab script with EEgLab functions, much more flexible and less time-consuming than using a GUI.
For my problem with linked mastoid reference (issue 145) for example, I use the solution kindly provided by Dounia, but it takes me roughly 1-2h to process 30 participants...

Best
Fabienne

hughherald, 2018-06-03:
Dear Fabienne,

I'm Gan Huang. 
The script is the important feature in Letswave7.
Recently, even drawing figures can be done by the automatically generated Matlab scripts. 
Really cool to me. You can try it.
https://github.com/NOCIONS/letswave7

Best,
Gan

fchetail, 2018-06-04:
Hi Gan, 
Thanks a lot, I’m going to have a look!
Best
Fabienne