NOCIONS/letswave6 #152: Letswave 5: invalid mex-file linux 16.04

simenhagen, 2018-08-16:
I am running linux 16.04.  While Letswave6 seems to work fine 6 works,  Letswave 5 does not.  When trying to import biosemi files,  I get the following error: 

" Invalid MEX-file '/home/simen/Documents/MATLAB/Letswave 5 new/external/fieldtrip/ft_getopt.mexa64': '" 

The same error appears when running Letswave on windows10.  

Do you know of an easy fix to this problem? 

Thanks,
Simen

amouraux, 2018-08-16:
Probably that the (old) fieldtrip mex file is problematic or missing. You may want to download it from the fieldtrip library, and replace it in the external/fieldtrip folder.

Best wishes,
André

simenhagen, 2018-08-16:
That could work.  Turns out it also works to use an older Matlab version (e.g., 2012). 

Thanks,
Simen