NOCIONS/letswave6 #154: L6: error in importing BDF file

YuFangYG, 2018-09-25:
Hi
I am running L6 on windown10. Letswave6 seems ok to load processed data when I am trying to import biosemi files, I get the following error:

" Undefined function 'ft_read_data' for input arguments of type 'char'.

Error in RLW_import_BDF (line 26)
dat=ft_read_data(filename);

Error in LW_import_BDF (line 86)
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_BDF(filename);

Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating uimenu Callback '"

I had updated the external/fieldtrip folder from fieldtrip library & Letswave6.  Do you know of a way to fix this problem?

Many thanks in advance,
Fang

amouraux, 2018-09-26:
I made a slight change and, hopefully it is now fixed?
If you stil get the error message, can you tell me what the name of your file is?

Best wishes,
André

YuFangYG, 2018-09-26:
Hi André,
Thanks for your reply. I'd redownloaded the L6 just now, sadly I still got the error msg as following while importing BDF file:  

"Invalid MEX-file
'C:\Users\MATLAB\letswave6-master\letswave6-master\external\fieldtrip\ft_getopt.mexw64':
The specified module could not be found.

Error in ft_read_data (line 115)
hdr             = ft_getopt(varargin, 'header');

Error in RLW_import_BDF (line 26)
dat=ft_read_data(filename);

Error in LW_import_BDF (line 86)
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_BDF(filename);

Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating uimenu Callback"

Many thanks!
Fang

amouraux, 2018-09-27:
Can you check whether that file exists (ft_getopt.mexw64)?
If so, could you send me the dataset to try to debug this (Fieldtrip) error?

André

yyffang, 2018-09-27:
Hi André,
In the L6, there is "ft_getopt" file (ft_getopt.mexw64). Here is the link for the dataset : 
https://www.dropbox.com/s/4shotlkg1jw0wor/TempDist_S05_08_11_13.bdf?dl=0

Please let me know if the link doesn't work, thanks a lot!
Fang

amouraux, 2018-09-29:
Link does not work?
André

https://www.dropbox.com/s/4shotlkg1jw0wor/TempDist_S05_08_11_13.bdf?dl=0

amouraux, 2018-10-02:
> May I have your email so I can send it to you? Thank you!

andre.mouraux@uclouvain.be<mailto:andre.mouraux@uclouvain.be>

amouraux, 2018-10-02:
I just downloaded the BDF file that you sent me, and I was able to import it without any trouble.
This must then be something specific to the OS you are using (I am using a Mac), i.e. a bug in the mex file for the Windows platform.

I will try tomorrow on a PC.

André

On 2 Oct 2018, at 09:55, André Mouraux <andre.mouraux@uclouvain.be<mailto:andre.mouraux@uclouvain.be>> wrote:

andre.mouraux@uclouvain.be<mailto:andre.mouraux@uclouvain.be>