NOCIONS/letswave6 #157: Anova with permutation testing

lgoudman, 2018-10-31:
Dear,

I’m currently using Letswave 6 to compare six merged datasets with the ANOVA with permutation testing. We have a between subject effect (3 types of pathology) and a within subject effect (2 test locations). I’m able to run the first permutation, however there is an error afterwards: 
“Index exceeds matrix dimensions.
Error in RLW_ANOVA (line 181)
                datasets2(j).data(i,:,:,:,:,:)=datasets(b(j)).data(i,:,:,:,:,:);
                
Error in LW_ANOVA (line 82)
        [out_datasets(1),out_datasets(2),message_string,cluster_distribution]=RLW_ANOVA(datasets,configuration.parameters.factor_data,configuration.parameters.table_data,'enable_parallel',configuration.parameters.enable_parallel,'alpha',configuration.parameters.alpha,'permutation',configuration.parameters.permutation,'num_permutations',configuration.parameters.num_permutations,'cluster_statistic',configuration.parameters.cluster_statistic,'cluster_threshold',configuration.parameters.cluster_threshold);”
Am I constructing this in a wrong way? We are not able to run the analysis with permutation testing. 
					W: location	B: patho
Location 1	Patho 1			1		1
		Patho 2			1		2
		Patho 3			1		3
Location 2 	Patho 1			2		1
		Patho 2			2		2
		Patho 3			2		3

Thank you for your help. 

Lisa

amouraux, 2018-11-01:
Could you send me a screenshot of how you configure the ANOVA?

Best wishes,
André

lgoudman, 2018-11-01:
Dear,

Below you can find the configuration parameters. 

Thank you

Lisa

lgoudman, 2018-11-01:
Dear,

Below you can find the configuration of the ANOVA. 

Thank you. 

Lisa 

[image]

amouraux, 2018-11-01:
Dear Lisa,

Seems correctly configured. There must be a small bug in the function. Could you send me the merged_epochs files so that I can test and debug?

André

[anova]<https://user-images.githubusercontent.com/40005710/47840158-af9e3400-ddb5-11e8-9b64-2953f2821da2.jpg>

lgoudman, 2018-11-01:
Dear,

Thank you for having a look at this. 

This is the link towards the files: https://www.dropbox.com/sh/wcbne5a5tpzcxhd/AAAiYjjRbKPdBhFDmQ3f7OoCa?dl=0

Thank you. 

Lisa