NOCIONS/letswave6 #158: F and p values interpretation of ANOVA with more than 2 datasets with permutation testing

fnedez, 2018-11-12:
Hi,

I'm using letswave 6. I have a 2X2 design. 
I run an ANOVA with more than two datasets with 500 permutations.
When I open the F and p values, in the epoch I have number 1 which I assume should be the no-corrected values of F or p (it depents which one I selected), then I have number 2, then numer 3 and number 4. 
What each number should indicate?

best
Barbara

dmulders, 2018-11-12:
Hello Barbara,

You're right for the first epoch of the two files (p and F values). The meaning of each epoch is the following:
- epoch 1: uncorrected p or F values
- epoch 2: the thresholded p or F values (within the significant clusters following the permutation testing)
- epoch 3: the significance (percentile value) of each cluster, based on the distribution of all the cluster values following permutation.
- epoch 4: the percentile value of each cluster (as epoch 3), but based on the distribution of the max cluster values.

I hope this helps you!

Best,
Dounia

fnedez, 2018-11-15:
Ok thank you! Now it is clear. It helps a lot!

I have another question. 
When I'm looking at the thresholded p or F values (epoch 2), in some channel the graphic show an orizzontal line without any pick or wave and at the same time I hear a sound and Matlab board shows this message:

Error using ylim (line 31)
Limits must be a 2-element vector of increasing numeric values.

Error in CGLW_multi_viewer>update_graph (line 259)
ylim(allaxis(1),[str2num(get(handles.yaxis_min_edit,'String'))
str2num(get(handles.yaxis_max_edit,'String'))]);

Error in CGLW_multi_viewer>channel_listbox_Callback (line 526)
update_graph(handles);

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in CGLW_multi_viewer (line 24)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)CGLW_multi_viewer('channel_listbox_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.

Do you know what it could mean?

Thanks in advance,
Barbara

dmulders, 2018-11-15:
I guess that this could be due to the fact that the considered epoch is constant along time (and therefore the auto lower and upper y limits are equal). Did you try to display this kind of "problematic epoch" together with another one which does not raise an error, eg epoch 2 for a different channel? Does the problem persist?

Best,
Dounia

fnedez, 2018-11-15:
Matlab shows this message only when there is any significant effect. That is, when there is a significant effect in a specific channel for example between 200 and 300 millisec, there isn't any message. But when there is the orizzontal line on zero, Matlab shows that message.  

Best,
Barbara

fnedez, 2018-11-15:
upps sorry "Matlab shows this message only when there isn't any significant effect."

dmulders, 2018-11-15:
So indeed it's due to the automatic y-limits that are set based on the data that you visualize. You can try to change the y-limits in the GUI in the "y-axis" frame (without ticking "auto").

fnedez, 2018-11-20:
Yes! You are right!

May I ask you another thing! I'm new with Letswave :-)
In the epoch 2 I can see the permutated results of F and p values along the Time.
So for example I can see if there is a significative effect between 200 and 300 millsec for a specifical channel, like Cz.

However If I want see the F and the p values of a more general ANOVA in which I have my experimental conditions and I add two topographical factors, like Anterior Posterior and Left and Right Hemisphere groupping together 4 channels for each group (e.g., Left Anterior, Left Posterior, Right Antherior and Right Posterior), how can I do that?

best
Barbara