NOCIONS/letswave6 #162: Butterworth filter zero phase

valebrown, 2018-11-26:
Dear all, 
I'm using Letswave 6.  Which option do I have to select to use a bandpass zero-phase Butterworth filter?

Thank you in advance, 
Valentina

dmulders, 2018-11-27:
Dear Valentina,

You only have to choose the frequency band, all the filters that you apply in letswave are by default zero-phase.

Best,
Dounia

valebrown, 2018-11-27:
Dear Dounia, 
thank you very much for your answer! Can I kindly ask you what does it mean the option "filter order"?

Thanks again, 
best, 
valentina

dmulders, 2018-11-27:
With pleasure! The filter order is the number of poles of the transfer function. It controls the decrease rate of the filter gain outside the pass band (increasing it will increase the steepness of the filter sides in the frequency domain, at the expense of increasing the border effects). You can choose it based on the filter transfer function, or you can test different values and compare the obtained signals (and their Fourier transforms). A typical choice is 4, but it can depend on the application and the frequency band you want to isolate.

Best,
Dounia

EVale, 2018-11-27:
Valebrown,

if you want to visually simulate and  appreciate the impact of different
filter orders you can try ERPlab that has got a nice interface for
filtering purposes. You can see also differences between IIR and FIR.

Best

e

valebrown, 2018-11-27:
Thank you very much Dounia and Evale!

closed 2018-11-27 by dmulders