NOCIONS/letswave6 #163: error when plotting topographies

LenaStroh, 2018-11-27:
Hi,
I get the following error when I try to draw 3D headplots via the scalp map SSEP series creator:

**Index exceeds matrix dimensions.

Error in CLW_headplot (line 42)
vector=vector(find(tp==1));

Error in GLW_figure_scalpmap_SSEP_series>headplot_btn_Callback (line 620)
        CLW_headplot(vector2,dataset.header,'maplimits',[cmin cmax]);

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in GLW_figure_scalpmap_SSEP_series (line 24)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)GLW_figure_scalpmap_SSEP_series('headplot_btn_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.**

I use letswave6 on ubuntu
Plotting 2-D headplots works. Also, when I inspect the 3D headplot in the waveform multiviewer window, it seems to work fine. Any suggestions on how to fix his?

Best,
Lena

amouraux, 2018-11-28:
Hello, you first need to “Assign electrode coordinates” (in Edit Electrodes)

LenaStroh, 2018-11-28:
Dear André,
Thank you for your reply. I assigned the electrode coordinates right at the
beginning:
[image: grafik.png]
also, when I check the header files, the chanlocs seems to be assigned:
[image: grafik.png]
Best,
Lena

-- 
*Anna-Lena Stroh*

University of Hamburg
Biological Psychology and Neuropsychology
Von-Melle-Park 11
20146 Hamburg
Germany

Phone: +49 (0)40 42838 2620
E-Mail: anna-lena.stroh@uni-hamburg.de
<https://webmail.rrz.uni-hamburg.de/imp/message.php?Horde=aae5d4cab7959ba355b448fc750b2955&mailbox=INBOX%2FGesendet&index=153#>